gbionics/idyntree #19: Clean up

traversaro, 2015-02-06:
A clean up is currently undergoing in `development` branch (v0.3) to facilitate development, maintenance and dissemination (and to finally merge the new "floating position/velocity/acceleration/torque" logic. 

* [ ] Get a basic example (compiled out of the superbuild and of iDynTree source tree) to work 
* [ ] Get the codyco-superbuild to build and work with v0.3 (check that all test pass)
* [ ] Get the walkman superbuild to build and check that all tests are passing
* [ ] Restore the iDyn / iDynTree consistency checks, use them also for checking urdf export. 
* [ ] Correctly compile and install useful tools:
 * [ ] urdf2dh 
 * [ ] par2urdf
 * [ ] urdf_get_mass

traversaro, 2015-02-10:
Restored the iDyn/iDynTree consistency checks, there was a bug in assuming that the ft sensor frame was the one of parent link instead of the child link.

francesco-romano, 2015-02-10:
- move all inclusions to .cpp files and use forward declarations
- transform all intra-library header inclusions to use quotes instead of angular brackets
- all intra-library inclusions should be "direct", i.e. #include "myHeader.h" vs #include "folder/myHeader.h"

traversaro, 2015-02-11:
@francesco-romano (not worth doing that for things in the yarp directory, ok for everything else).

traversaro, 2015-02-11:
@francesco-romano namespace of what? C++ namespace?

francesco-romano, 2015-02-11:
@traversaro yes (and also some folder restructuring).
e.g.
`kdl_codyco` => `idyntree/core`

anyway I saw you already did some work on regressors, etc..

traversaro, 2015-02-11:
Yes, most of the libraries are already in a namespace, we just need to appropriately change the name of the namespace.

traversaro, 2015-11-10:
Update: most of the headers now use forward declarations when possible. 
All code in the library should be contained in the `iDynTree` namespace. 
Using additional namespaces have been discarded for being to complex and to match the package structure of Matlab and Python bindings. 

Legacy code ( former kdl_codyco, kdl_format_io) will not be updated because it will be deprecated by the pure-iDynTree (and thus with native bindings) implementations.

closed 2015-11-10 by traversaro