gbionics/idyntree #20: [Test] Writing tests

francesco-romano, 2015-02-10:
We should add more tests. 
This should be done also in https://github.com/robotology-playground/yarp-wholebodyinterface and https://github.com/robotology-playground/wholebodyinterface.

cc
@traversaro @claudia-lat

traversaro, 2015-05-04:
It is nice to notice that given that now we have bindings, it is also possible to write tests in scripting languages such as Lua or Python, and execute them in a unified infrastructure using RTF.

traversaro, 2015-06-28:
Also Matlab tests would be nice, to prevent regressions like https://github.com/robotology-playground/idyntree/issues/53 . 

Ref (even if only available from Matlab R2013a) : http://it.mathworks.com/help/matlab/matlab-unit-test-framework.html

traversaro, 2015-11-10:
We know have ~38 tests files in idyntree. 
More can always be added, especially covering the Matlab bindings, but I think we can consider the issue as closed.

closed 2015-11-10 by traversaro