gbionics/idyntree #33: [Core] Wrapping std::string as in YARP ?

traversaro, 2015-05-08:
Apparently, SWIG is not really static linking friendly.. thus even on Windows we should properly export DLL. 

However in a lot of methods (for example in the API describing an instance of a sensor) we want to expose a function that returns (in a clearly not-real time safe fashion, unless you use a O(1) allocator) a string, as in 
~~~
std::string getName();
~~~ 

The problem is that apparently [1] returning std containers in DLL is not supported. You should then find an alternative solution.

# Solution 1: Use C-style interfaces  
A possible C-Style interface is 
~~~
void getName(char * name)
~~~
I would avoid using this solution, because it would be really cumbersome and error prone for the C++ user (and for SWIG bindings) to manually deal with char array buffers. 

# Solution 2: Wrap std::string as YARP does 
It is not really clear to me, but I guess that having a class similar to `yarp::os::ConstString` should solve this problem. This would probably cost some performance overhead when dealing with strings,  but given that in performance-intensive loops you should not deal with strings anyway, I guess it is acceptable. 

[1] : http://stackoverflow.com/questions/3564985/returning-stdstring-stdlist-from-dll

traversaro, 2015-05-08:
For the time being, I would go for:
# Solution 4: Build dynamics, and wait until things stop working
Perhaps in recent version of Visual Studio there are some magic that solve the problem of passing std containers across DLL. I would go to try to continue in the standard way, and do some binding testing with RTF to check that everything is working.

traversaro, 2015-09-03:
After a discussion with @lornat75 and reading https://support.microsoft.com/en-us/kb/168958 , I guess it should be safe to continue to use `std::string` in the interfaces.

closed 2015-09-03 by traversaro

traversaro, 2017-11-28:
It turns out that using STL-members in the interfaces and in the headers is not problematic after all, even in MSVC . For more details on this, see also this nice recap: https://docs.google.com/presentation/d/14Ags6mV-xQbN6OIfVOPcvDOW9M2WlgwGls-QYhZe2GQ/edit?usp=sharing .

By the way, it also turns out that is not so problematic to use static libraries with SWIG. 

cc @diegoferigo

ahoarau, 2017-11-28:
> but given that in performance-intensive loops you should not deal with strings anyway, I guess it is acceptable.

Oops ! shan’t I use idyntreeFunction( "name_of_frame") in RT loops ?

francesco-romano, 2017-11-28:
@ahoarau iDynTree has the counter part using indices for performance reasons.
The string version is slower (it search for the element every time)

diegoferigo, 2018-04-06:
@randaz81 @claudiofantacci @pattacini https://github.com/robotology/idyntree/issues/33#issuecomment-347439854