gbionics/idyntree #36: Sensors cleanup

traversaro, 2015-05-11:
There are some tasks necessary to finish the cleanup of the sensors iDynTree system.

Things necessary for this week, to produce the regressor bindings for @ommac and @DiegoRomeres  :
* [ ] Remove `Sensors` dependency on `KDL`, to enable the use of SWIG to produce bindings 
 * [ ] Substitute references to `KDL::Wrench` with `iDynTree::Wrench`. The code that uses `SensorsMeasurements` but is still using `KDL::Wrench` can be still supported using the functions in https://github.com/robotology-playground/idyntree/blob/master/core/include/kdl_codyco/KDLConversions.h  
 * [ ] Substitute references to `KDL::Frame` with `iDynTree::Transform` . 
 * [ ] Move the `simulateMeasurement` method, that dependes on a lot of stuff from the `core/kdl` library, in a separated files as a stand-alone function. 

(relatively) Low priority:
* [ ] Rename `SensorsTree` in something more meaningful 
* [ ] Add a Gyroscope sensor type & class (this will require adding a `AngularVelocity` class to `Core`.
* [ ] Add an Accelerometer sensor type & class (this will require adding an `LinearAcceleration` or `LinearProperAcceleration` (to be discussed) to `Core` 

cc @naveenoid can you look into this?

naveenoid, 2015-05-11:
im on it

naveenoid, 2015-05-14:
SensorsTree renamed to SensorsList in 600659ab077393e705e7b47feae3ff5349a70b28

traversaro, 2015-10-14:
@naveenoid  I was thinking about the solution we discussed yesterday and I was wondering how you can deal with different type of sensors (such as *internal* six-axis ft sensors as opposed to a six-axis force plate, in a sense a "sensor of external force") that have the "measurement type". 

In that case you cannot simply use the sensor type that you encoded for your measurement type, because there is not unequivocal mapping between the "measurement types" and the "sensor types". 

Perhaps leaving the SensorType enum in the interface is slightly more insecure (only runtime checking as opposed to compile) but easier (and **quicker**) to implement. : )

naveenoid, 2015-10-14:
> Assigned #36 <https://github.com/robotology/idyntree/issues/36> to

My idea was that each sensor type has a unique measurement type but its not
one to one. But let me think a bit after the first push. So we can debate
better.

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Cognitive Humanoids Lab,
Department of Robotics, Brain and Cognitive Sciences (RBCS),
Istituto Italiano di Tecnologia, Genova, Italy

naveenoid, 2015-10-14:
First version (incomplete still) is in https://github.com/robotology/idyntree/commit/793ad4efa87bc430cc9efdcfe019a8afb0d12b2f

traversaro, 2015-10-17:
@naveenoid any updates?

traversaro, 2015-11-18:
Solved in https://github.com/robotology/idyntree/pull/89 .

closed 2015-11-18 by traversaro