gbionics/idyntree #43: Problems in Compiling Matlab Bindings - Matlab R2015a

DiegoRomeres, 2015-06-09:
In order to compile bindings to iDynTree in Matlab (my Matlab version is Matlab R2015a) I tried to enable IDYNTREE_USES_MATLAB CMake options, but I got this error:

Add library idyntree
 src/DynTree.cppsrc/yarp_kdl.cppsrc/TorqueEstimationTree.cppsrc/iDyn2KDL.cppsrc
 /yarp_iDynTree.cppsrc/idyn2kdl_icub.cppinclude/iCub/iDynTree/DynTree.hinclude/
 iCub/iDynTree/yarp_kdl.hinclude/iCub/iDynTree/yarp_iDynTree.hinclude/iCub/iDyn
 Tree/iDyn2KDL.hinclude/iCub/iDynTree/TorqueEstimationTree.hinclude/iCub/iDynTr
 ee/idyn2kdl_icub.h

 CMake Error at
 /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108
 (message):
   Could NOT find Matlab (missing: Matlab_INCLUDE_DIRS Matlab_MEX_LIBRARY
   Matlab_MEX_EXTENSION) (found version "8.5")
 Call Stack (most recent call first):
   /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315
 (_FPHSA_FAILURE_MESSAGE)
   cmake/FindMatlab.cmake:1432 (find_package_handle_standard_args)
   bindings/matlab/CMakeLists.txt:35 (find_package)

so it seems that Matlab cannot be found. What can I try?

Diego

traversaro, 2015-06-09:
Mhhh... Is the WBI-Toolbox (https://github.com/robotology-playground/WBI-Toolbox) able to find your matlab without problems? Where is Matlab installed?

DiegoRomeres, 2015-06-10:
My Matlab is installed in 
/nfsd/opt/matlab2015a
which is outside my local home folder /home/Diego.

I'm going to install the WBI-Toolbox to see if it finds Matlab or not.

traversaro, 2015-06-10:
Ah, ok.. in this case you can set the CMake variable `MATLAB_ROOT_DIR` to points to the directory where matlab is installed.

DiegoRomeres, 2015-06-10:
Yes you were right, now the last part of the error disappeared. The only part left is:

 Add library idyntree
 src/DynTree.cppsrc/yarp_kdl.cppsrc/TorqueEstimationTree.cppsrc/iDyn2KDL.cppsrc
 /yarp_iDynTree.cppsrc/idyn2kdl_icub.cppinclude/iCub/iDynTree/DynTree.hinclude/
 iCub/iDynTree/yarp_kdl.hinclude/iCub/iDynTree/yarp_iDynTree.hinclude/iCub/iDyn
 Tree/iDyn2KDL.hinclude/iCub/iDynTree/TorqueEstimationTree.hinclude/iCub/iDynTr
 ee/idyn2kdl_icub.h

traversaro, 2015-06-10:
That is just a verbose print, we have to remove but is not harmful at all!

DiegoRomeres, 2015-06-10:
OK great. Thank you!

closed 2015-06-16 by traversaro