gbionics/idyntree #44: MATLAB bindings: iDynTreeMATLAB_wrap class not found

Ommac, 2015-06-09:
I have followed the simple steps outlined in https://github.com/robotology-playground/idyntree#bindings to generate the iDynTree MATLAB bindings.

Then, I have started MATLAB r2014b (UNIX version), and added iDynTree/bindings/matlab/* to the MATLAB path.

Later on, I have tried the two example scripts proposed in https://github.com/robotology-playground/idyntree/blob/master/doc/geometric_classes.md

The outputs were:

1)
```
Undefined function 'iDynTreeMATLAB_wrap' for input arguments of type 'double'.

Error in iDynTree.Position (line 9)
        tmp = iDynTreeMATLAB_wrap(80,'new_Position',varargin{:}); % FIXME

Error in test1 (line 4)
origin_A_B = iDynTree.Position(1,2,3)
 ```

2)
```
Undefined function 'iDynTreeMATLAB_wrap' for input arguments of type 'double'.

Error in iDynTree.Position (line 9)
        tmp = iDynTreeMATLAB_wrap(80,'new_Position',varargin{:}); % FIXME

Error in test2 (line 9)
origin_A_B = iDynTree.Position(1,2,3)
```

Long things short: class *iDynTreeMATLAB_wrap* does not seem to be defined.

**Note**: In folder iDynTree/bindings/matlab there is a file called *iDynTreeMATLAB_wrap.cxx*. I do not know what is it for, but I would expect to also have a MATLAB version of it, which I don't.

Maybe some options for generating the bindings are wrong?

Thanks.

traversaro, 2015-06-10:
The `iDynTreeMATLAB_wrap.cxx` file gets compiled in the `iDynTreeMATLAB_wrap.mexa64` file (the extension could change in different architectures/OSs ). 
This file (and the Matlab classes contained in ) are then installed in `${CMAKE_INSTALL_PREFIX}/mex/`. Consequently you should add to your path that directory.. but this is currently lacking from the README, because if you where using the `codyco-superbuild` you had already instructions on how to add that directory to Matlab path... thanks for spotting this lack!

traversaro, 2015-06-10:
I am not a great Matlab user, are this instructions https://github.com/robotology-playground/idyntree/commit/8aca61356e6be1a8fb1a97eb3cc9012727abe67f enough or we need to clarify the necessary steps?

Ommac, 2015-06-10:
Thanks Silvio,

I have found the mex64 file in the following way:

```
 'kammo@kammo-Latitude-E5540:~$ sudo find / -name 'iDynTreeMATLAB_wrap.mexa64'
[sudo] password for kammo: 
/home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/iDynTreeMATLAB_wrap.mexa64
```

The directory in which it has been placed seems different from the ones you suggested in https://github.com/robotology-playground/idyntree/commit/8aca61356e6be1a8fb1a97eb3cc9012727abe67f

Test 1 works correctly

Test 2 makes matlab crash. The crash dump is reported below

```

------------------------------------------------------------------------
              abort() detected at Wed Jun 10 17:19:58 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding     : Disabled
  Current Visual     : None
  Default Encoding   : UTF-8
  GNU C Library      : 2.19 stable
  MATLAB Architecture: glnxa64
  MATLAB Root        : /usr/local/MATLAB/R2014b
  MATLAB Version     : 8.4.0.150421 (R2014b)
  Operating System   : Linux 3.13.0-37-generic #64-Ubuntu SMP Mon Sep 22 21:28:38 UTC 2014 x86_64
  Processor ID       : x86 Family 6 Model 69 Stepping 1, GenuineIntel
  Software OpenGL    : 0
  Virtual Machine    : Java 1.7.0_11-b21 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System      : No active display

Fault Count: 1

Abnormal termination:
abort()

Register State (from fault):
  RAX = 0000000000000000  RBX = 00007f9b7c024000
  RCX = ffffffffffffffff  RDX = 0000000000000006
  RSP = 00007f9c0bff9d58  RBP = 00007f9c24c483d0
  RSI = 0000000000001b33  RDI = 0000000000001b02

   R8 = fefefefefefefeff   R9 = fefefefefefefeff
  R10 = 0000000000000008  R11 = 0000000000000202
  R12 = 00007f9b638d2faf  R13 = 00007f9b638d2fc0
  R14 = 00007f9c0bffa8a0  R15 = 00007f9c0bffa8a0

  RIP = 00007f9c24afdcc9  EFL = 0000000000000202

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007f9c24afdcc9                    /lib/x86_64-linux-gnu/libc.so.6+00224457 gsignal+00000057
[  1] 0x00007f9c24b010d8                    /lib/x86_64-linux-gnu/libc.so.6+00237784 abort+00000328
[  2] 0x00007f9c24af6b86                    /lib/x86_64-linux-gnu/libc.so.6+00195462
[  3] 0x00007f9c24af6c32                    /lib/x86_64-linux-gnu/libc.so.6+00195634
[  4] 0x00007f9b638c87b9 /home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00440249
[  5] 0x00007f9b638c8003 /home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00438275 _ZN8iDynTree18TransformSemantics15check_transformERKS0_RKNS_17PositionSemanticsE+00000085
[  6] 0x00007f9b638c3f0b /home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00421643 _ZN8iDynTree9Transform9transformERKS0_RKNS_8PositionE+00000073
[  7] 0x00007f9b638c4219 /home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00422425 _ZNK8iDynTree9TransformmlERKNS_8PositionE+00000043
[  8] 0x00007f9b63b1eaa7 /home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/iDynTreeMATLAB_wrap.mexa64+00232103 _Z30_wrap_Transform_mtimes__SWIG_1iPP11mxArray_tagiS1_+00000437
[  9] 0x00007f9b63b1f2ce /home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/iDynTreeMATLAB_wrap.mexa64+00234190 _Z22_wrap_Transform_mtimesiPP11mxArray_tagiS1_+00000397
[ 10] 0x00007f9b63b2494f /home/kammo/Repos/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/iDynTreeMATLAB_wrap.mexa64+00256335 mexFunction+00019025
[ 11] 0x00007f9c1b694c0a     /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so+00150538 mexRunMexFile+00000090
[ 12] 0x00007f9c1b6915c4     /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so+00136644
[... 44 lines cut ...]
[ 57] 0x00007f9c2678100f /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02523151
[ 58] 0x00007f9c2678117c /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02523516
[ 59] 0x00007f9c2677d07f /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02506879
[ 60] 0x00007f9c267824b5 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02528437
[ 61] 0x00007f9c267828e7 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02529511
[ 62] 0x00007f9c26782fc0 /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so+02531264 _Z25svWS_ProcessPendingEventsiib+00000080
[ 63] 0x00007f9c25fbf248   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00635464
[ 64] 0x00007f9c25fbf564   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00636260
[ 65] 0x00007f9c25fabcdd   /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmcr.so+00556253
[ 66] 0x00007f9c24e95182              /lib/x86_64-linux-gnu/libpthread.so.0+00033154
[ 67] 0x00007f9c24bc200d                    /lib/x86_64-linux-gnu/libc.so.6+01028109 clone+00000109

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

```

traversaro, 2015-06-10:
Mh... that is the one compiled in the build directory... did you run the install command on `iDynTree` repo, directly from the repo or by doing a `make` on codyco-superbuild ?

DiegoRomeres, 2015-06-10:
Anche io ho la stessa cosa, il  primo funziona e il secondo mi fa andare in
crash Matlab come a Raffaello:

MATLAB crash file:/home/auto/matlab_crash_dump.11829-1:

------------------------------------------------------------------------
              abort() detected at Wed Jun 10 17:49:52 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware
  Current Visual      : 0x20 (class 4, depth 24)
  Default Encoding    : UTF-8
  GNU C Library       : 2.19 stable
  Host Name           : pcautomatici09
  MATLAB Architecture : glnxa64
  MATLAB Root         : /nfsd/opt/matlab2015a
  MATLAB Version      : 8.5.0.197613 (R2015a)
  OpenGL              : hardware
  Operating System    : Linux 3.16.0-38-generic #52~14.04.1-Ubuntu SMP Fri
May 8 09:43:57 UTC 2015 x86_64
  Processor ID        : x86 Family 6 Model 60 Stepping 3, GenuineIntel
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java
HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : The X.Org Foundation (11501000), display :0

Fault Count: 1

Abnormal termination:
abort()

Register State (from fault):
  RAX = 0000000000000000  RBX = 00007fed24916000
  RCX = ffffffffffffffff  RDX = 0000000000000006
  RSP = 00007fedda2aeb38  RBP = 00007fedef4fe830
  RSI = 0000000000002e61  RDI = 0000000000002e35

   R8 = fefefefefefefeff   R9 = fefefeff092d6364
  R10 = 0000000000000008  R11 = 0000000000000206
  R12 = 00007feca9e4efdb  R13 = 00007feca9e4f000
  R14 = 00007fedda2af6d0  R15 = 00007fedda2af6d0

  RIP = 00007fedef3b4cc9  EFL = 0000000000000206

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007fedef3b4cc9
/lib/x86_64-linux-gnu/libc.so.6+00224457 gsignal+00000057
[  1] 0x00007fedef3b80d8
/lib/x86_64-linux-gnu/libc.so.6+00237784 abort+00000328
[  2] 0x00007fedef3adb86
/lib/x86_64-linux-gnu/libc.so.6+00195462
[  3] 0x00007fedef3adc32
/lib/x86_64-linux-gnu/libc.so.6+00195634
[  4] 0x00007feca9e447c1
/home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00440257
[  5] 0x00007feca9e4400b
/home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00438283
_ZN8iDynTree18TransformSemantics15check_transformERKS0_RKNS_17PositionSemanticsE+00000085
[  6] 0x00007feca9e3ff13
/home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00421651
_ZN8iDynTree9Transform9transformERKS0_RKNS_8PositionE+00000073
[  7] 0x00007feca9e40221
/home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/libidyntree-core-exp.so+00422433
_ZNK8iDynTree9TransformmlERKNS_8PositionE+00000043
[  8] 0x00007fecaa09aa87
/home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/lib/x86_64-linux-gnu/iDynTreeMATLAB_wrap.mexa64+00232071
[... 128 lines cut ...]
_Z25svWS_ProcessPendingEventsiib+00000092
[ 65] 0x00007fedf08c6408
/nfsd/opt/matlab2015a/bin/glnxa64/libmwmcr.so+00660488
[ 66] 0x00007fedf08c6724
/nfsd/opt/matlab2015a/bin/glnxa64/libmwmcr.so+00661284
[ 67] 0x00007fedf08b2b0d
/nfsd/opt/matlab2015a/bin/glnxa64/libmwmcr.so+00580365
[ 68] 0x00007fedef74b182
/lib/x86_64-linux-gnu/libpthread.so.0+00033154
[ 69] 0x00007fedef47847d
/lib/x86_64-linux-gnu/libc.so.6+01025149 clone+00000109
[ 70] 0x0000000000000000
<unknown-module>+00000000

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

DiegoRomeres, 2015-06-10:
I got the same results as Raffaello, Test1 works and Test2 crushes Matlab.

I run the install command directly on the repo.

Ommac, 2015-06-10:
I have tried both ways.

1) I went to *codyco-superbuild/libraries/iDynTree/build*, issueda a *ccmake ..* and set the option *IDYNTREE_USES_MATLAB = ON*. Then I issued a *make* instruction.
I did not issue a *sudo make install*
There is no *{CMAKE_INSTALL_PREFIX}/mex* directory

2)
I assumed that SUPERBUILD_BUILD_DIRECTORY means CMAKE_INSTALL_PREFIX
I set  *CODYCO_USES_MATLAB = ON*
and left *CMAKE_INSTALL_PREFIX = /usr/local*
I issued a *make update-all* in *codyco-superbuild/build*
I did not issue a *sudo make install* cmd
There is no *SUPERBUILD_BUILD_DIRECTORY/install/mex* directory

DiegoRomeres, 2015-06-11:
I tryed both cases as well and the results doesn't change.

Running the code line by line I saw that the error occurs in the last line:
O_A_wrong = transform_B_A*O_B; 

I guess the problem is related to the method 'Inverse()' of 'iDynTree.Transform' since the previous code worked properly. The error that apperas in the Matlab command Window befor the crash is:

[ERROR] TransformSemantics :: apply : Mismatch between orientationFrame of first operand and coordinateFrame of second operand

traversaro, 2015-06-11:
Sorry guys, this is my fault. : ) 

Basically the error handling procedure is calling an `assert(false)` if you do something wrong, to exit the program.. but this is not desirable in bindings where if you exit the program you exit the all interpreter! 
This happens for example in : 
* https://github.com/robotology-playground/idyntree/blob/3d87d9b8c2fd3e863eb0cea5e1751ae7f5760059/core-experimental/src/PositionSemantics.cpp#L87
* https://github.com/robotology-playground/idyntree/blob/3d87d9b8c2fd3e863eb0cea5e1751ae7f5760059/core-experimental/src/RotationSemantics.cpp#L67

The easiest fix is to remove those assert(false), or compile the library in Release. 

Notice that this was mentioned in the tutorial: `Semantics check will prevent us from doing stupid things (this will result in an error)`

DiegoRomeres, 2015-06-12:
I tried to run the test example after commenting the assert(false) you indicated us, but the problem is not solved. 
At a first view I didn't find the correct one to remove, but I will check the file more accurately and I will be back to you later.

DiegoRomeres, 2015-06-15:
Commenting this 'assert(false)':

https://github.com/robotology-playground/idyntree/blob/3d87d9b8c2fd3e863eb0cea5e1751ae7f5760059/core-experimental/src/Utils.cpp#L33

Matlab doesn't crash any more in the second test of:
 https://github.com/robotology-playground/idyntree/blob/master/doc/geometric_classes.md

traversaro, 2015-06-16:
@nunoguedelha is working on an improved version of the semantics checking, we will push it  soon and it will solve the problem of Matlab crashing on semantic error.

traversaro, 2021-09-15:
Semantics was removed in https://github.com/robotology/idyntree/pull/704, so automatically this bug is fixed.

closed 2021-09-15 by traversaro