gbionics/idyntree #47: MATLAB bindings not compiled with codyco-superbuild

Ommac, 2015-06-22:
I am building iDynTree via codyco-superbuild, by setting *CODYCO_USES_MATLAB      ON*, and the MATLAB mex files are not generated as expected in *SUPERBUILD_BUILD_DIRECTORY/install/mex*.

What's the right way to configure it?

traversaro, 2015-06-22:
iDynTree matlab bindigs generated was removed from `CODYCO_USES_MATLAB` because there was some problems with old version of Matlab ( ) .
In https://github.com/robotology/codyco-superbuild/commit/8b59664ca47cdaa4612e7411935badc85e43ed9f I have added to the codyco superbuild a `IDYNTREE_USES_MATLAB` option to explicitly compile iDynTree with matlab bindings. 
However the easiest way to compile the iDynTree matlab bindings is to `cd` in the iDynTree build directory (`cd codyco-superbuild/build/libraries/iDynTree/`) and execute `ccmake .` there to enable the options that you want. 

For the iDynTree-superbuild I still have to add the appropriate cmake options.

Ommac, 2015-06-23:
I tried the simple way. It now generates the mex-files.

However, when I run this example: https://github.com/robotology-playground/idyntree/blob/master/examples/matlab/iCubLeftArmDynamicRegressors/iCubLeftArmFTRegressor.m, after adding the whole codyco-superbuild directory to the MATLAB path, I get the following error:

```
Invalid MEX-file
'/home/kammo/Repos/codyco-superbuild/build/install/mex/iDynTreeMATLAB_wrap.mexa64':
libidyntree-core-exp.so: cannot open shared object file: No such file or
directory

Error in iDynTree.DynamicsRegressorGenerator (line 8)
        tmp =
        iDynTreeMATLAB_wrap(276,'new_DynamicsRegressorGenerator',varargin{:});
        % FIXME

Error in parametric_example (line 6)
regrGen = iDynTree.DynamicsRegressorGenerator();
 ```

traversaro, 2015-06-23:
From codyco-superbuild can you try to run : 
~~~
ldd ./build/install/mex/iDynTreeMATLAB_wrap.mexa64 
~~~

Ommac, 2015-06-23:
```
kammo@kammo-Latitude-E5540:~/Repos/codyco-superbuild$ ldd ./build/install/mex/iDynTreeMATLAB_wrap.mexa64 
	linux-vdso.so.1 =>  (0x00007ffc4879b000)
	libidyntree-core-exp.so => not found
	libidyntree-sensors.so => not found
	libidyntree-regressors.so.0.3.5 => not found
	libmex.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so (0x00007ffaece4b000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ffaecb46000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ffaec840000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ffaec62a000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ffaec264000)
	libmwfl.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwfl.so (0x00007ffaebe69000)
	libut.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libut.so (0x00007ffaebbac000)
	libmx.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmx.so (0x00007ffaeb7e5000)
	libmwservices.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so (0x00007ffaeb1d0000)
	libmwmpath.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmpath.so (0x00007ffaeaf88000)
	libmwm_dispatcher.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_dispatcher.so (0x00007ffaeac88000)
	libmwMATLAB_res.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwMATLAB_res.so (0x00007ffaea612000)
	libmwresource_core.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwresource_core.so (0x00007ffaea410000)
	libmwi18n.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwi18n.so (0x00007ffaea136000)
	libmwcpp11compat.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwcpp11compat.so (0x00007ffae9f29000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ffae9d0b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ffaed322000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ffae9b06000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007ffae98cd000)
	libboost_chrono.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_chrono.so.1.49.0 (0x00007ffae96c5000)
	libboost_date_time.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_date_time.so.1.49.0 (0x00007ffae94b4000)
	libboost_filesystem.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_filesystem.so.1.49.0 (0x00007ffae9295000)
	libboost_regex.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_regex.so.1.49.0 (0x00007ffae8f7a000)
	libboost_serialization.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_serialization.so.1.49.0 (0x00007ffae8d01000)
	libboost_signals.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_signals.so.1.49.0 (0x00007ffae8ae9000)
	libboost_system.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_system.so.1.49.0 (0x00007ffae88e6000)
	libboost_thread.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_thread.so.1.49.0 (0x00007ffae86ca000)
	libboost_log.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_log.so.1.49.0 (0x00007ffae83f2000)
	libboost_log_setup.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_log_setup.so.1.49.0 (0x00007ffae7ee1000)
	libicudata.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicudata.so.52 (0x00007ffae6674000)
	libicuuc.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicuuc.so.52 (0x00007ffae62e9000)
	libicui18n.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicui18n.so.52 (0x00007ffae5ec6000)
	libicuio.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicuio.so.52 (0x00007ffae5cb8000)
	libunwind.so.8 => /usr/local/MATLAB/R2014b/bin/glnxa64/libunwind.so.8 (0x00007ffae5a9a000)
	libtbb.so.2 => /usr/local/MATLAB/R2014b/bin/glnxa64/libtbb.so.2 (0x00007ffae584f000)
	libtbbmalloc.so.2 => /usr/local/MATLAB/R2014b/bin/glnxa64/libtbbmalloc.so.2 (0x00007ffae5603000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007ffae53fb000)
	libexpat.so.1 => /usr/local/MATLAB/R2014b/bin/glnxa64/libexpat.so.1 (0x00007ffae51d2000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ffae4fb9000)
	libmwregexp.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwregexp.so (0x00007ffae4d70000)
	libmwmlutil.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmlutil.so (0x00007ffae46a0000)
	libmwsettingscore.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwsettingscore.so (0x00007ffae422c000)
	libmwms.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwms.so (0x00007ffae3c62000)
	libmwMATLAB_settings_res.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwMATLAB_settings_res.so (0x00007ffae3a40000)
	libmwnativedisplay.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwnativedisplay.so (0x00007ffae3837000)
	libmwopccore.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwopccore.so (0x00007ffae35d7000)
	libmwopcmodel.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwopcmodel.so (0x00007ffae3352000)
	libmwopczippackage.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwopczippackage.so (0x00007ffae3131000)
	libmwopcmwservices.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwopcmwservices.so (0x00007ffae2c4b000)
	libboost_iostreams.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_iostreams.so.1.49.0 (0x00007ffae2a36000)
	libncurses.so.5 => /lib/x86_64-linux-gnu/libncurses.so.5 (0x00007ffae2813000)
	libPocoFoundation.so.21 => /usr/local/MATLAB/R2014b/bin/glnxa64/libPocoFoundation.so.21 (0x00007ffae2470000)
	libPocoNet.so.21 => /usr/local/MATLAB/R2014b/bin/glnxa64/libPocoNet.so.21 (0x00007ffae2174000)
	libxerces-c-3.1.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libxerces-c-3.1.so (0x00007ffae1aef000)
	libmwxmlcore.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwxmlcore.so (0x00007ffae187a000)
	libavrocpp.so.1.7.5.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libavrocpp.so.1.7.5.0 (0x00007ffae159b000)
	libmwflnetwork.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwflnetwork.so (0x00007ffae12ba000)
	libminizip.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libminizip.so (0x00007ffae10ad000)
	libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007ffae0e84000)
	libboost_program_options.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_program_options.so.1.49.0 (0x00007ffae0c25000)
```

Ommac, 2015-06-23:
Note that *libidyntree-core-exp.so* is in  *codyco-superbuild/build/install/lib/x86_64-linux-gnu*, not in   *codyco-superbuild/build/install/lib*. Maybe this is why it is not found.

traversaro, 2015-06-23:
For some reason rpath is not properly working... as a temporary workaround, add the directory that contains `libidyntree-core-exp.so` to `LD_CONFIG_PATH` .

traversaro, 2015-06-23:
The `x86_64-linux-gnu` is problem neverless, a bug of CMake < 3.1  (see http://public.kitware.com/pipermail/cmake-developers/2014-January/009435.html) but it should not affect the functionality of the library if RPATH was correctly working.

Ommac, 2015-06-23:
Fixed by adding the directory to *LD_LIBRARY_PATH* (not to *LD_CONFIG_PATH*)

Ommac, 2015-06-28:
This problem keeps showing up.

Strangely enough, I have to manually issue a *source .profile* in order to get it to work, and then restart MATLAB. See below:

```
kammo@kammo-Latitude-E5540:~/Repos/codyco-superbuild$ ldd ./build/install/mex/iDynTreeMATLAB_wrap.mexa64 
	linux-vdso.so.1 =>  (0x00007fff21bc3000)
	libidyntree-core-exp.so => not found
	libidyntree-sensors.so => /home/kammo/Repos/codyco-superbuild/./build/install/mex/../lib/libidyntree-sensors.so (0x00007f15eb0cb000)
	libidyntree-regressors.so.0.3.5 => /home/kammo/Repos/codyco-superbuild/./build/install/mex/../lib/libidyntree-regressors.so.0.3.5 (0x00007f15eac78000)
	libmex.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmex.so (0x00007f15eaa49000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f15ea744000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f15ea43e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f15ea228000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f15e9e62000)
	libidyntree-core-exp.so => not found
	liborocos-kdl.so.1.2 => /home/kammo/Repos/codyco-superbuild/./build/install/mex/../lib/../lib/liborocos-kdl.so.1.2 (0x00007f15e9bdd000)
	libidyntree-core.so => not found
	libidyntree-modelio-urdf.so => not found
	libtinyxml.so.2.6.2 => /home/kammo/Repos/codyco-superbuild/./build/install/mex/../lib/../lib/libtinyxml.so.2.6.2 (0x00007f15e99c2000)
	libidyntree-core-exp.so => not found
	liburdfdom_world.so.0.3 => /home/kammo/Repos/codyco-superbuild/./build/install/mex/../lib/../lib/liburdfdom_world.so.0.3 (0x00007f15e9795000)
	libmwfl.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwfl.so (0x00007f15e939a000)
	libut.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libut.so (0x00007f15e90dd000)
	libmx.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmx.so (0x00007f15e8d16000)
	libmwservices.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwservices.so (0x00007f15e8701000)
	libmwmpath.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmpath.so (0x00007f15e84b9000)
	libmwm_dispatcher.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwm_dispatcher.so (0x00007f15e81b9000)
	libmwMATLAB_res.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwMATLAB_res.so (0x00007f15e7b43000)
	libmwresource_core.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwresource_core.so (0x00007f15e7941000)
	libmwi18n.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwi18n.so (0x00007f15e7667000)
	libmwcpp11compat.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwcpp11compat.so (0x00007f15e745a000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f15e723c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f15eb59b000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f15e7037000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f15e6dfe000)
	libboost_chrono.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_chrono.so.1.49.0 (0x00007f15e6bf6000)
	libboost_date_time.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_date_time.so.1.49.0 (0x00007f15e69e5000)
	libboost_filesystem.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_filesystem.so.1.49.0 (0x00007f15e67c6000)
	libboost_regex.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_regex.so.1.49.0 (0x00007f15e64aa000)
	libboost_serialization.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_serialization.so.1.49.0 (0x00007f15e6232000)
	libboost_signals.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_signals.so.1.49.0 (0x00007f15e601a000)
	libboost_system.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_system.so.1.49.0 (0x00007f15e5e16000)
	libboost_thread.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_thread.so.1.49.0 (0x00007f15e5bfb000)
	libboost_log.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_log.so.1.49.0 (0x00007f15e5923000)
	libboost_log_setup.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_log_setup.so.1.49.0 (0x00007f15e5411000)
	libicudata.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicudata.so.52 (0x00007f15e3ba5000)
	libicuuc.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicuuc.so.52 (0x00007f15e381a000)
	libicui18n.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicui18n.so.52 (0x00007f15e33f6000)
	libicuio.so.52 => /usr/local/MATLAB/R2014b/bin/glnxa64/libicuio.so.52 (0x00007f15e31e9000)
	libunwind.so.8 => /usr/local/MATLAB/R2014b/bin/glnxa64/libunwind.so.8 (0x00007f15e2fcb000)
	libtbb.so.2 => /usr/local/MATLAB/R2014b/bin/glnxa64/libtbb.so.2 (0x00007f15e2d7f000)
	libtbbmalloc.so.2 => /usr/local/MATLAB/R2014b/bin/glnxa64/libtbbmalloc.so.2 (0x00007f15e2b34000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f15e292b000)
	libexpat.so.1 => /usr/local/MATLAB/R2014b/bin/glnxa64/libexpat.so.1 (0x00007f15e2703000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f15e24e9000)
	libmwregexp.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwregexp.so (0x00007f15e22a1000)
	libmwmlutil.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwmlutil.so (0x00007f15e1bd0000)
	libmwsettingscore.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwsettingscore.so (0x00007f15e175c000)
	libmwms.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwms.so (0x00007f15e1193000)
	libmwMATLAB_settings_res.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwMATLAB_settings_res.so (0x00007f15e0f70000)
	libmwnativedisplay.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwnativedisplay.so (0x00007f15e0d67000)
[... cut ...]
	libavrocpp.so.1.7.5.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libavrocpp.so.1.7.5.0 (0x00007fe273c0c000)
	libmwflnetwork.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libmwflnetwork.so (0x00007fe27392b000)
	libminizip.so => /usr/local/MATLAB/R2014b/bin/glnxa64/libminizip.so (0x00007fe27371e000)
	libtinfo.so.5 => /lib/x86_64-linux-gnu/libtinfo.so.5 (0x00007fe2734f5000)
	libboost_program_options.so.1.49.0 => /usr/local/MATLAB/R2014b/bin/glnxa64/libboost_program_options.so.1.49.0 (0x00007fe273296000)
```

nunoguedelha, 2015-06-28:
I don't have this issue, but another one, also related to `iDynTreeMATLAB_wrap.mexmaci64` mex file. I'm building on MAC and did an update of codyco-superbuild, all components and *iDynTree*  2 days ago (commit `dbc2192`).
When setting `IDYNTREE_COMPILE_BINDINGS` ON, `IDYNTREE_USES_MATLAB` ON (without generating the bindings), iDynTree compilation would fail when linking target `iDynTreeMATLAB_wrap.mexmaci64`:

```
Undefined symbols for architecture x86_64:
  "_mxCreateCharArray_730", referenced from:
      SWIG_FromCharPtrAndSize(char const*, unsigned long) in iDynTreeMATLAB_wrap.o
  "_mxCreateDoubleMatrix_730", referenced from:
      iDynTree_MatrixDynSize_toMatlab(iDynTree::MatrixDynSize const*) in iDynTreeMATLAB_wrap.o
      iDynTree_VectorDynSize_toMatlab(iDynTree::VectorDynSize const*) in iDynTreeMATLAB_wrap.o
  "_mxCreateDoubleScalar", referenced from:
      SWIG_From_double(double) in iDynTreeMATLAB_wrap.o
  "_mxCreateLogicalScalar", referenced from:
      SWIG_Matlab_ConvertPtrAndOwn(mxArray_tag*, void**, swig_type_info*, int, int*) in iDynTreeMATLAB_wrap.o
      SWIG_Matlab_NewPointerObj(void*, swig_type_info*, int) in iDynTreeMATLAB_wrap.o
```

I realised that the these symbols are defined in the Matlab `mx` library, and this library is loaded by cake only if explicitely required. Actually it's not required in the current version of `$CODYCO_SUPERBUILD_ROOT/libraries/iDynTree/bindings/matlab/CMakeLists.txt`.

I added the `MX_LIBRARY` in the find_package call and it's now working. @traversaro has this issue been observed before? If yes, could you please review the fix? I just pushed the branch `fix/47/mxLibraryMissing`. Thanks!

DiegoRomeres, 2015-07-21:
I'm getting again similar problems (A couple of weeks ago, it was executing fine in my computer).

Executing the command:
origin_A_B = iDynTree.Position(1,2,3) 

I obtain the error:

Invalid MEX-file '/home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/bindings/matlab/iDynTreeMATLAB_wrap.mexa64':
/home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/bindings/matlab/iDynTreeMATLAB_wrap.mexa64: undefined symbol:
_ZN8iDynTree11PositionRawC1EPKdj

*** MATLAB has encountered a Linux operating system bug; this bug will eventually cause MATLAB to crash. Do not repeat the command which produced this
error, as doing so might cause an immediate crash. Save your work and restart MATLAB now.

Error in iDynTree.Position (line 9)
        tmp = iDynTreeMATLAB_wrap(145,'new_Position',varargin{:}); % FIXME

Error in Test1 (line 4)
origin_A_B = iDynTree.Position(1,2,3)

While compiling the example https://github.com/robotology-playground/idyntree/blob/master/examples/matlab/iCubLeftArmDynamicRegressors/iCubLeftArmFTRegressor.m

 Matlab crashes.

Has the file 'iDynTreeMATLAB_wrap.mexa64 ' been changed?

traversaro, 2015-07-21:
When you updated the library? Now?

DiegoRomeres, 2015-07-21:
yes

traversaro, 2015-07-21:
Should be fixed by https://github.com/robotology/idyntree/commit/34957c70ccfd76e8245b54e8041a74596997f65e .

closed 2015-07-21 by traversaro

iron76, 2015-07-24:
@DiegoRomeres can you please check if it is indeed solved?

DiegoRomeres, 2015-07-24:
Sorry I didn't reply, it's fixed! It works fine!