gbionics/idyntree #50: Python bindings, No rule to make target `/usr/lib/python3.2/config/libpython3.2.so', needed by `lib/python/_iDynTree.so'

andreadelprete, 2015-06-23:
Hi,

I'm trying to compile the python bindings of iDynTree using the iDynTree-superbuild.
I updated the superbuild (git pull), updated all the repos (make update-all), re-built (cmake+make) and everything worked fine.
Then I went to build/lib/iDynTree to check that the flag IDYN_TREE_USES_PYTHON was ON, and it was, but it didn't generate the python bindings (in build/install/lib there is no directory python).
Then I tried compiling directly inside build/lib/iDynTree and it looks like it's trying to compile the python bindings, but it fails with this error:
```
make[2]: *** No rule to make target `/usr/lib/python3.2/config/libpython3.2.so', needed by `lib/python/_iDynTree.so'.  Stop.
```
even if I specified through the variable IDYN_TREE_USE_PYTHON_VERSION that I want to use python 2.7. Any idea what is going wrong?

Thanks,
Andrea

traversaro, 2015-06-23:
Ubuntu 12.04 ?

andreadelprete, 2015-06-23:
Yes, sorry I forgot to mention that!

traversaro, 2015-06-24:
There were some errors due to the copy&paste of cmake code from YARP:
https://github.com/robotology-playground/idyntree/commit/d1f6ec4de70faae6b093a24e3d5acbd377eac151
Can you try now? (I have also renamed the `IDYNTREE_USE_PYTHON_VERSION` to `IDYNTREE_USES_PYTHON_VERSION` for consistency).

andreadelprete, 2015-06-24:
I tried, now it seems it is no longer compiling iDynTree because there is no directory /build/libraries anymore! Is it possible that it found the installed version of iDynTree and so it doesn't compile it??

traversaro, 2015-06-24:
If another one is installed in the system, that could happen, but it should not be possible that the superbuild finds its own installed version. 
Anyway you can force the superbuild to download and build its own iDynTree (ignoring the one installed) invoking cmake with the option `USE_SYSTEM_IDYNTREE` set to OFF:
~~~
cmake -DUSE_SYSTEM_IDYNTREE=OFF .
~~~

andreadelprete, 2015-06-25:
Ok, I managed to make the super-build compile iDynTree, but I got the same error as before:
```
[100%] Building CXX object bindings/python/CMakeFiles/_iDynTree_python.dir/__/iDynTreePYTHON_wrap.cxx.o
make[2]: *** No rule to make target `/usr/lib/python3.2/config/libpython3.2.so', needed by `lib/python/_iDynTree.so'.  Stop.
make[1]: *** [bindings/python/CMakeFiles/_iDynTree_python.dir/all] Error 2
'''

traversaro, 2015-06-25:
Just to be sure: you updated the iDynTree repository through the superbuild command `make update-all` ?

andreadelprete, 2015-06-25:
I think I did, anyway I did it again now (just in case) and I get compilation errors:
```
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp: In member function ‘bool iDynTree::Regressors::DynamicsRegressorGenerator::loadRobotAndSensorsModelFromString(const string&, const string&)’:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:132:34: error: ‘sensorsTreeFromURDFString’ is not a member of ‘kdl_format_io’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp: In member function ‘bool iDynTree::Regressors::DynamicsRegressorGenerator::setRobotState(const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::Twist&)’:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:380:30: error: ‘Identity’ is not a member of ‘iDynTree::TransformRaw’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:381:27: error: ‘Zero’ is not a member of ‘iDynTree::SpatialMotionVectorRaw’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:382:31: error: ‘Zero’ is not a member of ‘iDynTree::SpatialMotionVectorRaw’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp: In member function ‘bool iDynTree::Regressors::DynamicsRegressorGenerator::setRobotState(const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::Transform&, const iDynTree::Twist&, const iDynTree::Twist&, const iDynTree::Twist&)’:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:410:49: error: no match for ‘operator*’ in ‘base_R_world * base_velocity’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:410:49: note: candidates are:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:89:18: note: iDynTree::Rotation iDynTree::Rotation::operator*(const iDynTree::Rotation&) const
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:89:18: note:   no known conversion for argument 1 from ‘const iDynTree::Twist’ to ‘const iDynTree::Rotation&’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:91:18: note: iDynTree::Position iDynTree::Rotation::operator*(const iDynTree::Position&) const
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:91:18: note:   no known conversion for argument 1 from ‘const iDynTree::Twist’ to ‘const iDynTree::Position&’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:411:63: error: no match for ‘operator*’ in ‘base_R_world * base_acceleration’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:411:63: note: candidates are:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:89:18: note: iDynTree::Rotation iDynTree::Rotation::operator*(const iDynTree::Rotation&) const
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:89:18: note:   no known conversion for argument 1 from ‘const iDynTree::Twist’ to ‘const iDynTree::Rotation&’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:91:18: note: iDynTree::Position iDynTree::Rotation::operator*(const iDynTree::Position&) const
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:91:18: note:   no known conversion for argument 1 from ‘const iDynTree::Twist’ to ‘const iDynTree::Position&’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:412:63: error: no match for ‘operator*’ in ‘base_R_world * world_gravity’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:412:63: note: candidates are:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:89:18: note: iDynTree::Rotation iDynTree::Rotation::operator*(const iDynTree::Rotation&) const
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:89:18: note:   no known conversion for argument 1 from ‘const iDynTree::Twist’ to ‘const iDynTree::Rotation&’
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:91:18: note: iDynTree::Position iDynTree::Rotation::operator*(const iDynTree::Position&) const
/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include/iDynTree/Core/Rotation.h:91:18: note:   no known conversion for argument 1 from ‘const iDynTree::Twist’ to ‘const iDynTree::Position&’
```

traversaro, 2015-06-25:
It is probably a bug on conflicting headers between the installed version and the source. Can you remove the install directory and try again? (sorry)

andreadelprete, 2015-06-25:
I removed the installed version, I ran again cmake, then make, and it gives the same errors.

traversaro, 2015-06-25:
I set  up the Travis (that runs on 12.04 ) for iDynTree-superbuild and apparently is correctly building: https://travis-ci.org/robotology-playground/iDynTree-superbuild/builds/68311247

andreadelprete, 2015-06-25:
Yes, the problem does not appear in the superbuild, but only when building inside iDynTree directory, i.e. /build/libraries/iDynTree, which I have to do because the superbuild does not generate the python bindings.

traversaro, 2015-06-25:
Mh... There is defiantly something that is not working, but once you define the  `IDYNTREE_USES_PYTHON` option in the iDynTree project this is saved in the cache in the `build/libraries/iDynTree` directory, so it is used also when building with the superbuild.

andreadelprete, 2015-06-30:
Problem solved! The last problem was that I did not remove properly the installed version, I forgot to remove the kdl_format_io headers. Now it compiles the python bindings. Thanks for the help @traversaro

traversaro, 2015-06-30:
@andreadelprete I am now working the bindings for the part that we discussed together!

andreadelprete, 2015-09-18:
Hi @traversaro , are the python bindings for the "joint-torque" regressor now available?

traversaro, 2015-09-18:
Yes, I had a bit of a backlog of local commit that I forgot to push. 
Now the required function are available in the Dynamics Computation class (http://wiki.icub.org/codyco/dox/html/classiDynTree_1_1HighLevel_1_1DynamicsComputations.html#a0186bd50edf39c42c4f5897311c9f9fc)

andreadelprete, 2015-10-05:
Is the DynamicsComputations class available directly in python? It does not work f I do: 
```
from iDynTree import DynamicsComputations
```

traversaro, 2015-10-05:
Mhh strange.. the command `from iDynTree import DynamicsComputations` works fine for me.

andreadelprete, 2015-10-05:
```PYTHONPATH``` contains ```~/devel/hyq/src/iDynTree-superbuild/build/install/lib/python2.7/dist-packages/``` and LD_LIBRARY_PATH contains ```~/devel/hyq/src/iDynTree-superbuild/build/install/lib```. To be sure I have a clean build I erased the old one and created a new ```build``` directory. In the idyntree namespace in python I have a lot of things, but not DynamicsComputations:
```
'DynamicsRegressorGenerator', 'DynamicsRegressorGenerator_swigregister', 
'DynamicsRegressorParameter', 'DynamicsRegressorParameter_swigregister', 
'DynamicsRegressorParametersList', 'DynamicsRegressorParametersList_swigregister', 'IMatrix', 
'IMatrix_swigregister', 'IVector', 'IVector_swigregister', 'LINK_FIRST_MOMENT_OF_MASS_X', 
'LINK_FIRST_MOMENT_OF_MASS_Y', 'LINK_FIRST_MOMENT_OF_MASS_Z', 'LINK_MASS', 
'LINK_MOMENT_OF_INERTIA_XX', 'LINK_MOMENT_OF_INERTIA_XY', 
'LINK_MOMENT_OF_INERTIA_XZ', 'LINK_MOMENT_OF_INERTIA_YY', 
'LINK_MOMENT_OF_INERTIA_YZ', 'LINK_MOMENT_OF_INERTIA_ZZ', 'LINK_PARAM', 
'MatrixDynSize', 'MatrixDynSize_swigregister', 'NR_OF_SENSOR_TYPES', 'Position', 'PositionRaw', 
'PositionRaw_compose', 'PositionRaw_inverse', 'PositionRaw_swigregister', 'PositionSemantics', 
'PositionSemantics_compose', 'PositionSemantics_inverse', 'PositionSemantics_swigregister', 
'Position_compose', 'Position_inverse', 'Position_swigregister', 'Rotation', 'RotationRaw', 
'RotationRaw_Identity', 'RotationRaw_RPY', 'RotationRaw_RotX', 'RotationRaw_RotY', 
'RotationRaw_RotZ', 'RotationRaw_compose', 'RotationRaw_inverse2', 'RotationRaw_swigregister', 
'RotationSemantics', 'RotationSemantics_compose', 'RotationSemantics_inverse2', 
'RotationSemantics_swigregister', 'Rotation_compose', 'Rotation_inverse2', 'Rotation_swigregister', 
'SENSOR_FT_OFFSET_FORCE_X', 'SENSOR_FT_OFFSET_FORCE_Y', 
'SENSOR_FT_OFFSET_FORCE_Z', 'SENSOR_FT_OFFSET_TORQUE_X', 
'SENSOR_FT_OFFSET_TORQUE_Y', 'SENSOR_FT_OFFSET_TORQUE_Z', 
'SENSOR_FT_PARAM', 'SIX_AXIS_FORCE_TORQUE', 'Sensor', 'Sensor_swigregister', 
'SensorsList', 'SensorsList_swigregister', 'SensorsMeasurements', 
'SensorsMeasurements_swigregister', 'SixAxisForceTorqueSensor', 
'SixAxisForceTorqueSensor_swigregister', 'SpatialForceVectorRaw', 'SpatialForceVectorRaw_Zero', 
'SpatialForceVectorRaw_compose', 'SpatialForceVectorRaw_inverse', 
'SpatialForceVectorRaw_swigregister', 'SpatialMotionVectorRaw', 'SpatialMotionVectorRaw_Zero', 
'SpatialMotionVectorRaw_compose', 'SpatialMotionVectorRaw_inverse', 
'SpatialMotionVectorRaw_swigregister', 'Transform', 'TransformRaw', 'TransformRaw_Identity', 
'TransformRaw_compose', 'TransformRaw_inverse2', 'TransformRaw_swigregister', 
'TransformRaw_transform', 'TransformSemantics', 'TransformSemantics_check_compose', 
'TransformSemantics_check_inverse2', 'TransformSemantics_check_transform', 
'TransformSemantics_compose', 'TransformSemantics_inverse2', 
'TransformSemantics_swigregister', 'TransformSemantics_transform', 'Transform_compose', 
'Transform_inverse2', 'Transform_swigregister', 'Transform_transform', 'Twist', 'Twist_swigregister', 
'Vector6', 'Vector6_swigregister', 'VectorDynSize', 'VectorDynSize_swigregister', 'Wrench', 
'Wrench_swigregister'
```

traversaro, 2015-10-05:
It seems that iDynTree is not updated to the latest version. A `make update-all` in the build directory of the superbuild should update iDynTree and all its dependencies. After that a `make` should compile and install the latest version of iDynTree.

andreadelprete, 2015-10-06:
I think I already ran ```make update-all```. Anyway I did it again and now I got this compilation error:
```
[ 79%] Building CXX object regressors/CMakeFiles/idyntree-regressors.dir/src/DynamicsRegressorGenerator.cpp.o
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp: In member function ‘bool iDynTree::Regressors::DynamicsRegressorGenerator::setRobotState(const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::Twist&)’:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:398:27: error: ‘SpatialMotionVector’ has not been declared
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:399:31: error: ‘SpatialMotionVector’ has not been declared
```
I tried removing the installed headers and libraries, but nothing changed.

traversaro, 2015-10-07:
Yes for some reason is finding some old headers somewhere... 
Can you try to check if there are some spurious header on your machine with `sudo updatedb && locate DynamicsRegressorGenerator.h` ?

andreadelprete, 2015-10-07:
I tried it, but nothing was found. I also checked manually in ```/usr/include```, ```/usr/local/include``` but I found nothing. Then I checked the command used by cmake to compile DynamicsRegressorGenerator:
```
[ 79%] Building CXX object regressors/CMakeFiles/idyntree-regressors.dir/src/DynamicsRegressorGenerator.cpp.o
cd /home/adelpret/devel/hyq/src/iDynTree-superbuild/build/libraries/iDynTree/regressors && /usr/bin/c++   -DOROCOS_KDL_VERSION_MAJOR=1 -DOROCOS_KDL_VERSION_MINOR=3 -DOROCOS_KDL_VERSION_PATCH=0 -DYARP_NO_DEPRECATED_WARNINGS -Didyntree_regressors_EXPORTS -O3 -DNDEBUG -fPIC -I/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/include/kdl_codyco/regressors -I/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/include -isystem /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/core-experimental/include -isystem /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/model/include -isystem /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/core/include -isystem /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/sensors/include -isystem /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/model_io/urdf/include -isystem /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/model_io/symoro/include -isystem /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/model_io/iKin/include -isystem /usr/include/eigen3 -isystem /opt/ros/hydro/include -I/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/include -I/home/adelpret/devel/hyq/src/iDynTree-superbuild/build/install/include    -o CMakeFiles/idyntree-regressors.dir/src/DynamicsRegressorGenerator.cpp.o -c /home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp: In member function ‘bool iDynTree::Regressors::DynamicsRegressorGenerator::setRobotState(const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::Twist&)’:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:398:27: error: ‘SpatialMotionVector’ has not been declared
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:399:31: error: ‘SpatialMotionVector’ has not been declared
```
All include directories are inside the superbuild, apart from ```/usr/include/eigen3``` and ```/opt/ros/hydro/include```, which of course do not contain any iDynTree header. Do you know a way to figure out which header is the compiler finding?

traversaro, 2015-10-07:
Mhh I think by comment on outdated headers was simply wrong then, the compilation is failing for some other reason. 
@nunoguedelha what do you think of the error message ? 
~~~
[ 79%] Building CXX object regressors/CMakeFiles/idyntree-regressors.dir/src/DynamicsRegressorGenerator.cpp.o
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp: In member function ‘bool iDynTree::Regressors::DynamicsRegressorGenerator::setRobotState(const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::VectorDynSize&, const iDynTree::Twist&)’:
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:398:27: error: ‘SpatialMotionVector’ has not been declared
/home/adelpret/devel/hyq/src/iDynTree-superbuild/libraries/iDynTree/regressors/src/DynamicsRegressorGenerator.cpp:399:31: error: ‘SpatialMotionVector’ has not been declared
~~~

Note that the `SpatialMotionVector` is a CRTP class, so I wonder if @andreadelprete is not using some compiler that does not support some template feature. That would be surprising for me, because the travis infrastructure runs on Ubuntu 12.04 and everything is compiling fine there.

traversaro, 2015-10-07:
On Travis iDynTree is compiling fine on gcc 4.6.3 and clang 3.4 .

andreadelprete, 2015-10-07:
I checked that my compiler version is indeed 4.6.3

andreadelprete, 2015-10-08:
Taking a look at the code it seems to me that the problem is due to the inclusion of the wrong ```Twist.h``` header. There are two ```Twist.h``` headers in iDynTree, one in ```core``` and one in ```core-experimental```. For some reason in my system the header in ```core``` is installed and so included in ```DynamicsRegressorGenerator.cpp```, whereas (I guess) it should be the one in ```core-experimental``` (that includes ```SpatialMotionVector.h```) to be included.

traversaro, 2015-10-08:
Actually there is just only one `Twist.h` file in `core-experimental` that is installed in `iDynTree/Core/Twist.h` (to be future-proof, given that as you may imagine `core-experimental` will become `core`) . I have the impression then that the problem is that the compilation should take the header in the source tree, not the one installed. I have already tried to resolve this issue, but now I have a better understanding on what is going on, and so I hope I will be able to solve it.

traversaro, 2015-10-08:
In particular I find particularly fishy that source include directories are passed to compile with the `-isystem` option. That is definitly the first thing to solve.

traversaro, 2015-10-08:
Can you try if https://github.com/robotology/idyntree/commit/188524e44b7cd12450468ca5f06b518c5b70c530 solved something?

andreadelprete, 2015-10-08:
It works now! I had to clean the install directory and then it worked. Thanks for the help @traversaro !

closed 2015-10-08 by andreadelprete

gtournois, 2015-10-14:
Andrea, how did you remove the kdl_format_io headers?

gtournois, 2015-10-14:
And more important maybe, where can I find them?

gtournois, 2015-10-14:
I have this error:
<pre><code>
/usr/bin/ld: /usr/local/lib/libpython2.7.a(abstract.o): relocation R_X86_64_32S against `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython2.7.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/python/_iDynTree.so] Error 1
make[1]: *** [bindings/python/CMakeFiles/_iDynTree_python.dir/all] Error 2
</pre></code>

traversaro, 2015-10-14:
@gtournois on which platform? Feel free to open a new issue anyway. : )

gtournois, 2015-10-15:
Hi Sylvio, 
Just as Andrea I am trying to compile iDynTree with the Python bindings and my problem is more or less the same, therefore I decided to not open a new issue. I am running Ubuntu 14.04 and Python 2.7.1.

 I did the same as Andrea, I cloned the branche, then 
<code> cmake -DIDYN_TREE_USES_PYTHON=ON; make update-all;  make </code>
It compiled without errors, however there is no folder called Python in superbuild/build/install/lib/. 

I tried to compile directly in the build/libraries/iDynTree folder. There were no errors, but still no folder called python or bindings in the build/install/lib. 

So in the build directory I did: <code>cmake ../ -DUSE_SYSTEM_IDYNTREE=OFF -DIDYN_TREE_USES_PYTHON=ON</code>, again make update-all, make. Still no folder.

Then I changed in the CMakeCache.txt which is in build/libraries/iDynTree 
IDYNTREE_USES_PYTHON:BOOL=FALSE in IDYNTREE_USES_PYTHON:BOOL=TRUE. and obtained following error
<code> [100%] Building CXX object bindings/python/CMakeFiles/_iDynTree_python.dir/__/iDynTreePYTHON_wrap.cxx.o
Linking CXX shared module ../../lib/python/_iDynTree.so
/usr/bin/ld: /usr/local/lib/libpython2.7.a(abstract.o): relocation R_X86_64_32S against `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython2.7.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[5]: *** [lib/python/_iDynTree.so] Error 1
make[4]: *** [bindings/python/CMakeFiles/_iDynTree_python.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [libraries/iDynTree/CMakeFiles/YCMStamp/iDynTree-build] Error 2
make[1]: *** [CMakeFiles/iDynTree.dir/all] Error 2
make: *** [all] Error 2
</code>

Thanks!

traversaro, 2015-10-15:
Hi @gtournois , there are three issues: 
  * The iDynTree-superbuild does not support correctly the `IDYNTREE_USES_PYTHON` options, I need to correct it. 
  * The option is `IDYNTREE_USES_PYTHON`, not `IDYN_TREE_USES_PYTHON`. 
  * It appears that you have a installation of python not coming from deb packages, for which only static libraries are installed. (`/usr/local/lib/libpython2.7.a` is mentioned in the error, and `/usr/local` is the prefix where non-system packages are installed. The system python libraries can be found on Ubuntu 14.04 in `/usr/lib/x86_64-linux-gnu/libpython2.7.so` and `/usr/lib/x86_64-linux-gnu/libpython2.7.a`. 

To solve your problem (i.e. addressing the third bullet point) you can just specify which python installation to use by following the `FindPythonLibs` docs [1] : set the cmake variable `PYTHON_LIBRARY` to `/usr/lib/x86_64-linux-gnu/libpython2.7.so` and the `PYTHON_INCLUDE_DIR` to `/usr/include/python2.7/`. Anyway I would strongly suggest that you get rid of the non-system python install, that could cause several problems to you in the future. 

By the way, do you know that you can easily modify the CMake cache using the `ccmake` command?

[1]

gtournois, 2015-10-15:
It works! Thanks for the tips! 
I am relatively new to Ubuntu, compilers, makefiles, etc. Could you maybe explain what is the proper way to uninstall a non system program, in this case Python 2.7. 
Is it correct to do a system-wide install I have to do <code> sudo apt-get install packagename </code>?

traversaro, 2015-10-15:
Yes, that is the correct way for installing a system package.
 If you are relatively new to Ubuntu, compilers, makefiles, etc.. I wonder how that non-system python install was installed.. Perhaps it is not a clean Ubuntu setup, somebody else already used it?

gtournois, 2015-10-15:
Alright! It was a clean install. As you would do with Windows, I just downloaden zip-file from python.org and followed the build options in the readme file. I think I will keep on working with the current python version, until I have other problems. Thanks for your help!

traversaro, 2015-10-15:
Ah, you did it then! 
In *nix system usually python come preinstalled with the system, so it is better to use that one to avoid incompatibilities. To remove the one you installed if you are lucky it is sufficient to go in the build directory and try to type `sudo make uninstall`.