gbionics/idyntree #51: iDynTree is not finding boost include [windows 7]

nirosis21, 2015-06-24:
While compiling iDynTree with the codyco-superbuild I am getting the following error:

> 6>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\install\include\urdf_model/pose.h(52): fatal error C1083: Cannot open include file: 'boost/algorithm/string.hpp': No such file or directory 

It seems like boost is not properly found by cmake despite envirnoment variables (`BOOST_DIR`, `BOOST_LIBRARYDIR` and `BOOST_INCLUDEDIR`) were properly defined. This is also suggested by the inlude directories list:

> c:\Program Files\Microsoft Visual Studio 10.0\VC\bin\CL.exe /c /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\install\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\model_io\urdf\external\urdfdom\..\urdfdom_headers\urdf_model\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\model_io\urdf\external\urdfdom\..\urdfdom_headers\urdf_exception\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\model_io\urdf\external\urdfdom\..\urdfdom_headers\urdf_sensor\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\model_io\urdf\external\urdfdom\..\urdfdom_headers\urdf_world\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\model_io\urdf\external\urdfdom\..\urdfdom_headers\urdf_model_state\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\model_io\urdf\external\urdfdom\urdf_parser\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\core-experimental\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\core\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\sensors\include" /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\libraries\iDynTree\include" /Zi /nologo /W3 /WX- /Od /Ob0 /Oy- /D WIN32 /D _WINDOWS /D _DEBUG /D URDF_USE_PLAIN_POINTERS /D IDYNTREE_USE_INTERNAL_URDFDOM /D YARP_NO_DEPRECATED_WARNINGS /D OROCOS_KDL_VERSION_MAJOR=1 /D OROCOS_KDL_VERSION_MINOR=2 /D OROCOS_KDL_VERSION_PATCH=3 /D URDFDOM_DO_NOT_USE_CONSOLEBRIDGE /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /GR /Fo"iDynTree_priv_urdfdom_model.dir\Debug\\" /Fd"iDynTree_priv_urdfdom_model.dir\Debug\vc100.pdb" /Gd /TP /analyze- /errorReport:queue ..\..\..\..\..\..\..\..\libraries\iDynTree\model_io\urdf\external\urdfdom\urdf_parser\src\pose.cpp ..\..\..\..\..\..\..\..\libraries\iDynTree\model_io\urdf\external\urdfdom\urdf_parser\src\model.cpp ..\..\..\..\..\..\..\..\libraries\iDynTree\model_io\urdf\external\urdfdom\urdf_parser\src\link.cpp ..\..\..\..\..\..\..\..\libraries\iDynTree\model_io\urdf\external\urdfdom\urdf_parser\src\joint.cpp
6>    pose.cpp

Other projects (like urdfdom) properly configure boost (only after defining  `BOOST_DIR`, `BOOST_LIBRARYDIR` and `BOOST_INCLUDEDIR`):

> 29>  ClCompile:
> 29>    c:\Program Files\Microsoft Visual Studio 10.0\VC\bin\CL.exe /c /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\install\include" /IC:\boost /I"C:\Users\Bastien\Documents\GitHub\codyco-superbuild\external\urdfdom\urdf_parser\include" /nologo /W3 /WX- /O2 /Ob2 /Oy- /D WIN32 /D _WINDOWS /D NDEBUG /D URDFDOM_DO_NOT_USE_CONSOLEBRIDGE /D "CMAKE_INTDIR=\"Release\"" /D urdfdom_model_EXPORTS /D _WINDLL /D _MBCS /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /GR /Fo"urdfdom_model.dir\Release\\" /Fd"urdfdom_model.dir\Release\vc100.pdb" /Gd /TP /analyze- /errorReport:queue ..\..\..\..\external\urdfdom\urdf_parser\src\pose.cpp ..\..\..\..\external\urdfdom\urdf_parser\src\model.cpp ..\..\..\..\external\urdfdom\urdf_parser\src\link.cpp ..\..\..\..\external\urdfdom\urdf_parser\src\joint.cpp ..\..\..\..\external\urdfdom\urdf_parser\src\urdf_sensor.cpp ..\..\..\..\external\urdfdom\urdf_parser\src\urdf_model_state.cpp ..\..\..\..\external\urdfdom\urdf_parser\src\twist.cpp

traversaro, 2015-06-24:
iDynTree recently dropped the dependency on Boost (on Windows), but for an error on my side the necessary changes to support this were propagate to the codyco-superbuild. I now propagated this changes in https://github.com/robotology/codyco-superbuild/commit/7a4420662a3352026ba703dbe87a79c0e191cfe3 . 

What is happening here is that there is a conflict between the internal copy of urdfdom and the one found in the system, that is clearly a bug that we have to address. For the moment, I would suggest to update the codyco-superbuild, remove the Boost environmental variables, remove everything relative to `urdfdom` and `urdfdom_headers` in the build and install directory of the codyco-superbuild and clean che cache of the codyco-superbuild, tryng to replicate a clean build as much as possible. 

Alternatively, given that Boost, urdfdom and urdfdom_headers are correctly installed you can try to disable the `IDYNTREE_USE_INTERNAL_URDFDOM` option in the iDynTree. In that case iDynTree will not try to use its own urdfdom and urdfdom_headers, but it will instead find the system one as it does by default in *nix systems. 

To modify the options of a superbuild subproject, you can simply open in `cmake-gui` the subproject build directory, for example something like `/codyco-superbuild/build/libraries/iDynTree`. After that you can edit iDynTree CMake options as you do in any CMake project.

nirosis21, 2015-06-25:
Following the first option (the one you recommended: "I would suggest to update the codyco-superbuild ...") the compilation goes forward but now fails here:

> 2>..\..\..\..\libraries\iDynTree\regressors\src\DynamicsRegressorParameters.cpp(16): fatal error C1083: Cannot open include file: 'boost/concept_check.hpp': No such file or directory

traversaro, 2015-06-25:
Sorry, that header should not have been there : https://github.com/robotology-playground/idyntree/commit/6f0fa779f933e8d2c34effef680d5fc358a80f43 . 
If you update the superbuild (target `ALL_UPDATE` or something similar) everything should (hopefully) compile.

nirosis21, 2015-06-25:
Now the error is:

> 3>..\..\..\..\libraries\iDynTree\regressors\src\DynamicsRegressorParameters.cpp(71): error C2039: 'lower_bound' : is not a member of 'std' [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\libraries\iDynTree\regressors\idyntree-regressors.vcxproj]

traversaro, 2015-06-25:
Fixed in https://github.com/robotology-playground/idyntree/commit/96cadc6b5e1f5d4fde2ccfbede467c65b18b0c65 . Sorry, but we still don't have a continuous integration job running on Windows...

nirosis21, 2015-06-29:
Thanks for your help. 
Now I get that error: 

>3>..\..\..\..\libraries\iDynTree\regressors\src\DynamicsRegressorGenerator.cpp(342): warning C4018: '<' : signed/unsigned mismatch [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\libraries\iDynTree\regressors\idyntree-regressors.vcxproj]
3>c:\users\bastien\documents\github\codyco-superbuild\libraries\idyntree\regressors\src\dynamicsregressorgenerator.cpp(85): error C4716: 'iDynTree::Regressors::DynamicsRegressorGenerator::operator=' : must return a value [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\libraries\iDynTree\regressors\idyntree-regressors.vcxproj]

traversaro, 2015-06-29:
Thanks for reporting, it should be fixed by https://github.com/robotology-playground/idyntree/commit/ff239ffcf37c8e36d133e55afd489e6d0525cc8c .

nirosis21, 2015-06-29:
I have now a bunch of errors related to "wholeBodyReach"

>36>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\main\codyco-modules\src\modules\wholeBodyReach\include\wholeBodyReach/eiquadprog2011.hpp(166): warning C4101: 'j' : unreferenced local variable [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyReach\wholeBodyReach.vcxproj]
36>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\main\codyco-modules\src\modules\wholeBodyReach\include\wholeBodyReach/eiquadprog2011.hpp(485): warning C4101: 'i' : unreferenced local variable [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyReach\wholeBodyReach.vcxproj]
36>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\main\codyco-modules\src\modules\wholeBodyReach\include\wholeBodyReach/wbiMinJerkTasks.h(66): warning C4800: 'double' : forcing value to bool 'true' or 'false' (performance warning) [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyReach\wholeBodyReach.vcxproj]
36>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\main\codyco-modules\src\modules\wholeBodyReach\include\wholeBodyReach/wbiMinJerkTasks.h(127): error C2719: 'posDes': formal parameter with __declspec(align('8')) won't be aligned [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyReach\wholeBodyReach.vcxproj]
36>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\main\codyco-modules\src\modules\wholeBodyReach\include\wholeBodyReach/wbiMinJerkTasks.h(132): error C2719: 'ctrlPoint': formal parameter with __declspec(align('8')) won't be aligned [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyReach\wholeBodyReach.vcxproj]
36>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\main\codyco-modules\src\modules\wholeBodyReach\include\wholeBodyReach/wbiMinJerkTasks.h(191): error C2719: 'comDes': formal parameter with __declspec(align('8')) won't be aligned [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyReach\wholeBodyReach.vcxproj]
36>C:\Users\Bastien\Documents\GitHub\codyco-superbuild\main\codyco-modules\src\modules\wholeBodyReach\include\wholeBodyReach/wbiConstraints.h(87): error C2719: 'normalDir': formal parameter with __declspec(align('8')) won't be aligned 

and so on....

I also have this error:

> "C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyDynamicsTree\wholeBodyDynamicsTree.vcxproj" (cible par défaut) (20) ->
36>  (ClCompile cible) -> 
36>..\..\..\..\..\..\main\codyco-modules\src\modules\wholeBodyDynamicsTree\src\filters.cpp(233): error C2065: 'M_PI' : undeclared identifier [C:\Users\Bastien\Documents\GitHub\codyco-superbuild\build\main\codyco-modules\src\modules\wholeBodyDynamicsTree\wholeBodyDynamicsTree.vcxproj]

traversaro, 2017-12-06:
The latest of iDynTree does not use anymore boost, so this issue should not be a problem anymore.

closed 2017-12-06 by traversaro