gbionics/idyntree #53: [matlab] Solve problems with access to internal attribute by returning references

traversaro, 2015-06-26:
iDynTree makes an intensive use of method returning references to compose different classes
such as SensorsMeasurements or *Semantics. A recent commit in matlab swig broke this
behaviour, until we fix this issue we can rollback to the latest matlab swig version working ( jaeandersson/swig@e435cba  ). 

@nunoguedelha currently I am generating bindings with this swig-matlab commit : https://github.com/jaeandersson/swig/commit/e435cba21e33c664e8ee75f5f9bd17b245aeb0b7 . So if you checkout to that commit (`git checkout e435cba21e33c664e8ee75f5f9bd17b245aeb0b7`) in the swig directory we will be in synch. 

Upstream issue: https://github.com/jaeandersson/swig/issues/28

nunoguedelha, 2015-06-29:
Done. Thanks!

traversaro, 2015-07-01:
Ok, switched to commit 7aff6628508d3f148794b131dfbdfb2b4f6c7c60 on @nunoguedelha fork : https://github.com/nunoguedelha/swig/tree/addCoutRedirectionToMatlab .

traversaro, 2015-08-25:
The problem was solved upstream ( ) so now we are generating the binding with this commit of swig-matlab : https://github.com/jaeandersson/swig/commit/8b8ae6df4def16d1589c7d262e95c4247601f120 .

closed 2015-08-25 by traversaro