gbionics/idyntree #54: Not working method 'getDescriptionOfDegreesOfFreedom()' in class 'DynamicsRegressorGenerator'

DiegoRomeres, 2015-06-26:
I got the problem playing with the Matlab regressor example in:

https://github.com/robotology-playground/idyntree/tree/master/examples/matlab/iCubLeftArmDynamicRegressors

In the specifc case, calling:
regrGen.getDescriptionOfDegreesOfFreedom()

I got as result:
Empty string: 1-by-0

when instead there are 32 degree of freedom.

traversaro, 2015-06-26:
Mh.. for me it is working in the latest commit. Have you updated to the latest version?

DiegoRomeres, 2015-06-26:
yes I just checked, I updated it... @Ommac did you get the problem?

traversaro, 2015-06-28:
I started to see the same problem... it is really strange because sometimes it works sometime.. a print of my terminal:
~~~
>> regrGen.getDescriptionOfDegreesOfFreedom()

ans =

DOF Index: 0 Name: l_hip_pitch
DOF Index: 1 Name: l_hip_roll
DOF Index: 2 Name: l_hip_yaw
DOF Index: 3 Name: l_knee
DOF Index: 4 Name: l_ankle_pitch
DOF Index: 5 Name: l_ankle_roll
DOF Index: 6 Name: r_hip_pitch
DOF Index: 7 Name: r_hip_roll
DOF Index: 8 Name: r_hip_yaw
DOF Index: 9 Name: r_knee
DOF Index: 10 Name: r_ankle_pitch
DOF Index: 11 Name: r_ankle_roll
DOF Index: 12 Name: torso_pitch
DOF Index: 13 Name: torso_roll
DOF Index: 14 Name: torso_yaw
DOF Index: 15 Name: l_shoulder_pitch
DOF Index: 16 Name: l_shoulder_roll
DOF Index: 17 Name: l_shoulder_yaw
DOF Index: 18 Name: l_elbow
DOF Index: 19 Name: l_wrist_prosup
DOF Index: 20 Name: l_wrist_pitch
DOF Index: 21 Name: l_wrist_yaw
DOF Index: 22 Name: neck_pitch
DOF Index: 23 Name: neck_roll
DOF Index: 24 Name: neck_yaw
DOF Index: 25 Name: r_shoulder_pitch
DOF Index: 26 Name: r_shoulder_roll
DOF Index: 27 Name: r_shoulder_yaw
DOF Index: 28 Name: r_elbow
DOF Index: 29 Name: r_wrist_prosup
DOF Index: 30 Name: r_wrist_pitch
DOF Index: 31 Name: r_wrist_yaw

>> regrGen.getDescriptionOfDegreesOfFreedom()

ans =

   Empty string: 1-by-0
~~~

traversaro, 2015-06-28:
It should be fixed by https://github.com/robotology-playground/idyntree/commit/9a45a60e967142cf4aaad6980f6cd7a8c05b361a .

closed 2015-06-28 by traversaro