gbionics/idyntree #58: Dynamics computations feature elicitation

traversaro, 2015-07-01:
@prashanthr05 and @naveenoid are planning to use the new classes for spatial algebra. 
While we are finishing to implement them, I would ask them to do the following: 
 * Try to compile the matlab bindings, following the instruction on the README
 * Try to play with the bindings, following the tutorial at https://github.com/robotology/idyntree/blob/master/doc/geometric_classes.md 
  * Comment this issue with desired features
  * Write some "unit-test" in matlab, for checking that  the new datastructure are sound.

traversaro, 2015-07-01:
I pushed a first backbone of test infrastructure for matlab in https://github.com/traversaro/rbcs-report-latex-template 
Some notes: 
 * I did not use the matlab unit testing infrastructure because I want to be compatibile with Matlab 2012
 * To run the test you simply run the `runTest` script, provided that you added iDynTree to your matlab path. 
 * To add a new test take inspiration from `testSumPos.m` and then call it from the main `runTest.m` 

I still need some CMake magic to integrate this with the `CTest` infrastructure (i.e. call this script with `make test`), but for now it will be enough. 

As always improvements are more then welcome.

traversaro, 2015-07-02:
Please not the assertion function use exit, so it is better to call the testing code with the following command to the command line:
~~~
matlab -nodesktop -r "runiDynTreeTests"
~~~

traversaro, 2015-07-02:
As part of https://github.com/robotology/idyntree/pull/59 , I have introduced the concepts of Spatial Inertia, Spatial Acceleration , Spatial Momentum and extended Wrench and Twist with some classical operation of spatial algebra. 
A brief explanation:

~~~
A_T_B = iDynTree.Transform()
p_B = iDynTree.Position()

% The point p_B expressed in B
% is transformed in p_A , considering both the orientation
% and the translation
p_A = A_T_B*p_B 

v_B  = iDynTree.Twist()
% The twist v_B expressed in B (i.e. orientation of B and 
% expressed with respect to the origin of B)
% is transformed in v_A (i.e. expressed with the orientation of A and 
% expressed with respect to the origin of A)
% This is basically equivalent to a multiplication for the adjoint matrix,
% i.e. in featherstone terms: v_a = A_X_B*v_b
v_A = A_T_B*v_B

% The spatial acceleration a_B expressed in B (i.e. orientation of B and 
% expressed with respect to the origin of B)
% is transformed in a_A (i.e. expressed with the orientation of A and 
% expressed with respect to the origin of A)
% This is basically equivalent to a multiplication for the adjoint matrix,
% i.e. in featherstone terms: a_a = A_X_B*a_b
a_B  = iDynTree.Twist()
a_A = A_T_B*a_B

% The spatial momentum h_B expressed in B (i.e. orientation of B and 
% expressed with respect to the origin of B)
% is transformed in h_A (i.e. expressed with the orientation of A and 
% expressed with respect to the origin of A)
% This is basically equivalent to a multiplication for the adjoint matrix of wrench,
% i.e. in featherstone terms: h_a = A_X^{*}_B*h_b
h_B  = iDynTree.SpatialMomentum()
h_A = A_T_B*h_B

% The wrench f_B expressed in B (i.e. orientation of B and 
% expressed with respect to the origin of B)
% is transformed in f_A (i.e. expressed with the orientation of A and 
% expressed with respect to the origin of A)
% This is basically equivalent to a multiplication for the adjoint matrix of wrench,
% i.e. in featherstone terms: f_a = A_X^{*}_B*f_b
h_B  = iDynTree.Wrench()
f_A = A_T_B*f_B

% The inertia I_B expressed in B (i.e. orientation of B and 
% the center of mass and inertia matrix expressed with respect to the origin of B)
% is transformed in f_A (i.e. expressed with the orientation of A and 
%  the center of mass and inertia matrix with respect to the origin of A)
% This is basically equivalent to the transformation for the spatial inertia,
% i.e. in featherstone terms: I_A = A_X^{*}_B*I_B*B_X_A
I_B = iDynTree.SpatialInertia()
I_A = A_T_B*I_B

% The inertia maps velocities to momentum and acceleration to wrench
I = iDynTree.SpatialInertia()
a = iDynTree.SpatialAcc()
v  = iDynTree.Twist()
f = I*a
h = I*v

% The spatial cross products maps twists to acceleration and momentum to wrench:
t = iDynTree.Twist()
v  = iDynTree.Twist()
h = iDynTree.SpatialMomentum()
a  = t*v
f    = t*h 
~~~

nunoguedelha, 2015-07-28:
Hi Silvio, I've pushed my dev branch `feature/29/AddSemChecksToSpatialVectors` as is. It still doesn't compile. The split of Spatial vector classes into linear + angular vector classes is done. I just have to add the semantics in these atomic classes, and it will be spread through inheritance to the derived and composed classes (Spatial vector classes).
Many of these classes are templates, except for those at the bottom of the inheritance tree. There are no template functions (as far as I remember I always avoided to use them).
I will be back to IIT on monday 10/08.
I will work on the code again since wednesday 05/08.

traversaro, 2015-10-21:
@prashanthr05 and @naveenoid have been already using the `DynamicsComputations` class for some time now, I guess then it fits their needs. For new feature requests new issues can always been opened.

closed 2015-10-21 by traversaro