gbionics/idyntree #61: Crash when using new semantics on Matlab at exit

francesco-romano, 2015-07-10:
Matlab crashes on quit

This is the log:
```

------------------------------------------------------------------------
       Segmentation violation detected at Fri Jul 10 14:24:12 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding     : Disabled
  Current Visual     : Quartz
  Default Encoding   : US-ASCII
  MATLAB Architecture: maci64
  MATLAB Root        : /Applications/MATLAB_R2014b.app
  MATLAB Version     : 8.4.0.150421 (R2014b)
  Operating System   : Darwin 14.4.0 Darwin Kernel Version 14.4.0: Thu May 28 11:35:04 PDT 2015; root:xnu-2782.30.5~1/RELEASE_X86_64 x86_64
  Processor ID       : x86 Family 6 Model 69 Stepping 1, GenuineIntel
  Software OpenGL    : 0
  Virtual Machine    : Java 1.7.0_55-b13 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System      : Quartz

Fault Count: 1

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 00007fa052ee9140  RBX = 0000000000000000
  RCX = 000000010e5aa4b0  RDX = 00000001014c1e2d
  RSP = 000000010e5aac60  RBP = fb00e6f00d267d26
  RSI = 000000010e5aa484  RDI = 0000000000000001

   R8 = 0000000000000001   R9 = 0000000000000000
  R10 = 0000000000000051  R11 = 0000000000000001
  R12 = 000000010e5aa4b0  R13 = 0000000118a6628f
  R14 = 00007fa0601bd7f0  R15 = 000000010e5aac60

  RIP = 000000010e5aa620  RFL = 0000000118a8f3d4

   CS = 0000000000000000   FS = 000000010e5aab98   GS = 0000000000000000

Stack Trace (from fault):
[  0] 0x0000000101244e64 /Applications/MATLAB_R2014b.app/bin/maci64/libmwfl.dylib+00028260 _ZN2fl4diag15stacktrace_base7captureERKNS0_14thread_contextEm+00000052
[  1] 0x0000000101248a2a /Applications/MATLAB_R2014b.app/bin/maci64/libmwfl.dylib+00043562 _ZN2fl4test17terminate_handledEv+00000906
[  2] 0x0000000101248477 /Applications/MATLAB_R2014b.app/bin/maci64/libmwfl.dylib+00042103 _ZN2fl4diag13terminate_logEPKcPK17__darwin_ucontext+00000119
[  3] 0x0000000100c9ab7a /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00387962 _Z32mnRunPathDependentInitializationv+00003146
[  4] 0x0000000100c9ae58 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00388696 _Z32mnRunPathDependentInitializationv+00003880
[  5] 0x0000000100c98733 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00378675 mnFatalSignalHandler+00000275
[  6] 0x00007fff82d74f1a           /usr/lib/system/libsystem_platform.dylib+00020250 _sigtramp+00000026
[  7] 0x00007fa0600e58f0                                   <unknown-module>+00000000
[  8] 0x0000000118a8f3d4 /Users/makaveli/Projects/src/codyco-superbuild/build/install/mex/iDynTreeMATLAB_wrap.mexmaci64+00246740 mexFunction+00005204
[  9] 0x0000000105d1beeb /Applications/MATLAB_R2014b.app/bin/maci64/libmex.dylib+00069355 mexRunMexFile+00000091
[ 10] 0x0000000105d17f41 /Applications/MATLAB_R2014b.app/bin/maci64/libmex.dylib+00053057 _ZN7Mfh_mex30runMexFileWithSignalProtectionEiPP11mxArray_tagiS2_+00000113
[ 11] 0x0000000105d18130 /Applications/MATLAB_R2014b.app/bin/maci64/libmex.dylib+00053552 _ZN7Mfh_mex13dispatch_fileEiPP11mxArray_tagiS2_+00000304
[ 12] 0x0000000105da5c28 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_dispatcher.dylib+00400424 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000600
[ 13] 0x0000000106705d62 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01523042 _Z19inDispatchFromStackiPKcii+00001650
[ 14] 0x00000001066a7dd8 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01138136 inPcodeWalker+00022552
[ 15] 0x00000001066a4917 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01124631 inPcodeWalker+00009047
[ 16] 0x000000010669ffb7 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01105847 inInterPcode+00000935
[ 17] 0x00000001066a11c6 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01110470 _Z26inExecuteMFunctionOrScriptP6Mfh_mpb+00000886
[ 18] 0x00000001067431f1 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01774065 _Z10inRunMfileiPP11mxArray_tagiS1_P6Mfh_mpP15inWorkSpace_tag+00002337
[ 19] 0x0000000105da5c28 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_dispatcher.dylib+00400424 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000600
[ 20] 0x0000000105f89e8e /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcos.dylib+01093262 _ZNK4mcos17MScopedMethodInfo6invokeEPKNS_9COSClientERiPNS_8COSValueEiPKS5_+00000606
[ 21] 0x0000000105fa4387 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcos.dylib+01201031 _ZNK4mcos13MDestructorMI6invokeEPKNS_9COSClientERiPNS_8COSValueEiPKS5_+00000551
[ 22] 0x0000000105e8c8c3 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcos.dylib+00055491 _ZN4mcos9COSClassI14callDestructorENS_15COSInterfacePtrE+00000179
[ 23] 0x0000000105e98299 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcos.dylib+00103065 _ZN4mcos9COSClassI15destroyInstanceENS_15COSInterfacePtrE+00000073
[ 24] 0x0000000105edd28f /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcos.dylib+00385679 _ZN4mcos13COSInterfaceI13destroyObjectEv+00000095
[ 25] 0x0000000105ed45e2 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcos.dylib+00349666 _ZN4mcos13COSInterfaceI10destroyRefEv+00000258
[ 26] 0x0000000105ed88d0 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcos.dylib+00366800 _ZN4mcos15COSInterfacePtr7releaseEv+00000112
[... 41 lines cut ...]
[ 68] 0x00000001066e648b /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01393803 _Z23inEvalPcodeHeaderToWordP15_memory_contextiPP11mxArray_tagP12_pcodeheaderP6Mfh_mpj+00000219
[ 69] 0x0000000106698f4b /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01077067 _Z25in_local_call_with_setjmpIN5boost3_bi6bind_tIvPFvP15_memory_contextPiPP11mxArray_tagP12_pcodeheaderjENS1_5list5INS1_5valueIS4_EENS0_3argILi1EEENSG_ILi2EEENSE_ISA_EENSE_IiEEEEEEEN6mlutil14cmddistributor17inExecutionStatusET_S5_S8_b+00000171
[ 70] 0x0000000106694221 /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01057313 _Z12inEvalStringP15_memory_contextRKSbItSt11char_traitsItESaItEE8EvalTypeiPP11mxArray_tag12inDebugCheckP12_pcodeheaderPib+00002769
[ 71] 0x00000001066947cd /Applications/MATLAB_R2014b.app/bin/maci64/libmwm_interpreter.dylib+01058765 _ZN12InterpBridge22EvalCmdWithLocalReturnERKSbItSt11char_traitsItESaItEEPibb+00000141
[ 72] 0x0000000105caf11d /Applications/MATLAB_R2014b.app/bin/maci64/libmwbridge.dylib+00061725 _Z28evalCommandWithLongjmpSafetyRKSbItSt11char_traitsItESaItEE+00000093
[ 73] 0x0000000105caf804 /Applications/MATLAB_R2014b.app/bin/maci64/libmwbridge.dylib+00063492 _Z8mnParserv+00000772
[ 74] 0x0000000100c82832 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00288818 _ZN11mcrInstance30mnParser_on_interpreter_threadEv+00000034
[ 75] 0x0000000100c5d0d1 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00135377 _ZN3mcr7runtime17InterpreterThread4Impl22TypedInvocationRequestIvE29doNotRethrowMxArrayExceptionsERKN5boost8functionIFvvEEE+00000033
[ 76] 0x0000000100c5dd89 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00138633 _ZN5boost6detail11task_objectIvNS_3_bi6bind_tIvPFvRKNS_8functionIFvvEEEENS2_5list1INS2_5valueIS6_EEEEEEE6do_runEv+00000025
[ 77] 0x0000000100c5d586 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00136582 _ZN5boost6detail9task_baseIvE3runEv+00000166
[ 78] 0x0000000100c5d377 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00136055 _ZN3mcr7runtime17InterpreterThread4Impl22TypedInvocationRequestIvE3runEv+00000023
[ 79] 0x0000000100c58f90 /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00118672 _ZN3mcr7runtime17InterpreterThread4Impl26invocation_request_handlerEl+00000032
[ 80] 0x000000010079a005 /Applications/MATLAB_R2014b.app/bin/maci64/libmwservices.dylib+00196613 _ZN10eventqueue18UserEventQueueImpl5flushEv+00001205
[ 81] 0x0000000107167a65 /Applications/MATLAB_R2014b.app/bin/maci64/libmwuix.dylib+00043621 _Z25isHandleUserEventCallbackPFvPvE+00000501
[ 82] 0x0000000100869a45 /Applications/MATLAB_R2014b.app/bin/maci64/libmwservices.dylib+01047109 _Z21svWS_ProcessCallbacksv+00016693
[ 83] 0x0000000100865152 /Applications/MATLAB_R2014b.app/bin/maci64/libmwservices.dylib+01028434 _Z25svWS_ProcessPendingEventsiib+00000834
[ 84] 0x0000000100864e5e /Applications/MATLAB_R2014b.app/bin/maci64/libmwservices.dylib+01027678 _Z25svWS_ProcessPendingEventsiib+00000078
[... cut ...]
[ 86] 0x0000000100c5949f /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00119967 _ZN3mcr7runtime17InterpreterThread4Impl3runERKN5boost10shared_ptrIS2_EEPNS2_12init_contextE+00000335
[ 87] 0x0000000100c3fd1b /Applications/MATLAB_R2014b.app/bin/maci64/libmwmcr.dylib+00015643 _Z26run_init_and_handle_eventsPv+00000059
[ 88] 0x00007fff8b4e4268            /usr/lib/system/libsystem_pthread.dylib+00012904 _pthread_body+00000131
[ 89] 0x00007fff8b4e41e5            /usr/lib/system/libsystem_pthread.dylib+00012773 _pthread_body+00000000
[ 90] 0x00007fff8b4e241d            /usr/lib/system/libsystem_pthread.dylib+00005149 thread_start+00000013

```

traversaro, 2015-07-10:
Master? I am afraid that was solved in matlab-swig in https://github.com/jaeandersson/swig/issues/22 , but we can't use it until we solve the regression in https://github.com/jaeandersson/swig/issues/28 .

traversaro, 2015-08-25:
This was solved by solving https://github.com/robotology/idyntree/issues/53 .

closed 2015-08-25 by traversaro