gbionics/idyntree #64: Compilation error:  undefined reference iDynTree::SpatialForceVectorRaw::operator-() const

rlober, 2015-07-21:
Just tried compiling the head of iDynTree and I get the following error:

```
-- Build files have been written to: /home/ryan/Code/iCub_Software/codyco-superbuild/build/libraries/iDynTree
[ 43%] Performing build step for 'iDynTree'
[ 27%] Built target idyntree-core-exp
[ 60%] Built target idyntree-core
[ 62%] Built target idyntree-sensors
[ 66%] Built target idyntree-modelio-urdf
[ 71%] Built target expression-parser
[ 73%] Built target idyntree-modelio-symoro
[ 75%] Built target idyntree-modelio-ikin
[ 90%] Built target idyntree-regressors
[ 97%] Built target idyntree-yarp
Linking CXX executable par2urdf
../sensors/libidyntree-sensors.so: undefined reference to `iDynTree::SpatialForceVectorRaw::operator-() const'
collect2: ld returned 1 exit status
make[5]: *** [tools/par2urdf] Error 1
make[4]: *** [tools/CMakeFiles/par2urdf.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [libraries/iDynTree/CMakeFiles/YCMStamp/iDynTree-build] Error 2
make[1]: *** [CMakeFiles/iDynTree.dir/all] Error 2
make: *** [all] Error 2
ryan@ISIR2014057:~/Code/iCub_Software/codyco-superbuild/build$ 
```

traversaro, 2015-07-21:
Can you try now? It should have been fixed by https://github.com/robotology/idyntree/commit/f96091fd2dabc06527cd5ca9e67462420cffcc7f .

rlober, 2015-07-21:
Fixed! :+1: @traversaro

closed 2015-07-21 by rlober