gbionics/idyntree #70: [Bindings] Missing / useful feature

francesco-romano, 2015-08-24:
Issue to be filled as a todo - feature request for matlab bindings (maybe also c++ / other bindings)

- [ ] cross operator (which returns a matrix)

traversaro, 2015-08-24:
Something like: 
~~~
namespace iDynTree
{
     Matrix3x3 cross(Vector3 &vec);
}
~~~
?

francesco-romano, 2015-08-24:
yes but in 6D

nunoguedelha, 2015-08-25:
Well, we could have a 3D and 6D cross product, as for the other operators once my next push is approved (if approved) and merged.
Also we should avoid static or local functions, since they prevent us from using virtuality if needed. Maybe it was already your idea @traversaro.
So I suggest we define `cross`, for a further use like :

~~~
LinearForceVector3 linearV; 
Twist twist;
Matrix3x3 skewLinearV;
Matrix6x6 skewTwist;

skewLinearV = linearV.cross();
skewTwist   = twist.cross();
~~~