gbionics/idyntree #74: [Bindings] [MATLAB] Errors are not printed out in Matlab terminal

traversaro, 2015-09-07:
@nunoguedelha They used to be printed in the Matlab terminal, right?

nunoguedelha, 2015-09-08:
Well, it depends on how you launch Matlab. I think that if you do it from the shell terminal, error print out in that terminal. We had also observed some issues related to SWIG, and used the fix from Joel Andersson (`[MATLAB] #7 Redirecting std::cout and std::cerr to mexPrintf`) . But since he introduced a flag for activating this redirection ([MATLAB] #7 No redirect of std::cout and std::cerr by default):
    `bool redirectoutput;`
Have you set this?
I haven't checked the print feature for a while.

traversaro, 2016-10-26:
In Linux and OS X, I got the errors. In Windows I was not able to see them.