gbionics/idyntree #76: [Bindings] [MATLAB] save/load not supported

traversaro, 2015-09-15:
save/load is not supported. 
As a first step, overload saveobj / loadobj to produce a clear error message explaining the problem (this probably should be done at swig level). 
cc @naveenoid @francesco-romano

naveenoid, 2015-09-17:
dumb solution is to implement some simple matlab functions (save_idyntree,
load_idyntree) which converts to a matlab matrix or structure and saves and
then reads from a matlab matrix and structure and regenerates idynTree. I
can implement one quickly if you think its a viable option.

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Cognitive Humanoids Lab,
Department of Robotics, Brain and Cognitive Sciences (RBCS),
Istituto Italiano di Tecnologia, Genova, Italy

traversaro, 2015-09-17:
Yes, this can be done in a transparent overloading the `saveobj` and `loadobj` methods. 
We need to understand how to do this in a way that it integrates with the swig code generation process (we cannot directly modify the generated code).
Upstream issue for reference: https://github.com/jaeandersson/swig/issues/59

DiegoRomeres, 2015-09-30:
Any news on this topic? At the moment when I save in Matlab I have to take care of deleting or converting in matlab matrices all the the iDyntree structures, right?
because I did not do that and now when I try to load my data I have the following error message:

Warning: Variable 'parametricEstimator' originally saved as a
IncrementalParametricEstimator cannot be instantiated as an object and will be read in
as a uint32.

------------------------------------------------------------------------
       Segmentation violation detected at Wed Sep 30 11:19:24 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware 
  Current Visual      : 0x20 (class 4, depth 24)
  Default Encoding    : UTF-8
  GNU C Library       : 2.19 stable
  Host Name           : pcautomatici09
  MATLAB Architecture : glnxa64
  MATLAB Root         : /nfsd/opt/matlab2015b
  MATLAB Version      : 8.6.0.267246 (R2015b)
  OpenGL              : hardware
  Operating System    : Linux 3.16.0-50-generic #66~14.04.1-Ubuntu SMP Thu Sep 10 17:05:00 UTC 2015 x86_64
  Processor ID        : x86 Family 6 Model 60 Stepping 3, GenuineIntel
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : The X.Org Foundation (11501000), display :0

Fault Count: 2

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 0000000000000000  RBX = 0000000000000001
  RCX = 00007f667b71a4c0  RDX = 00007f67b8000038
  RSP = 00007f67bf24c3d0  RBP = 00007f67bf24cc38
  RSI = 00000000ffffffff  RDI = 0000000000000000

   R8 = 0000000000000000   R9 = 00007f66baa5cd30
  R10 = 00007f67bf249e00  R11 = 00007f6791ac0790
  R12 = 00007f67bf24cc38  R13 = 00007f667b69a390
  R14 = 0000000000000001  R15 = 00007f67bf24cb60

  RIP = 00007f668ca546f0  EFL = 0000000000010246

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007f668ca546f0 /home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/bindings/matlab/iDynTreeMATLAB_wrap.mexa64+00288496 _Z24_wrap_VectorDynSize_sizeiPP11mxArray_tagiS1_+00000064
[  1] 0x00007f668cacdf65 /home/auto/MEGA/iCub/Software/codyco-superbuild/libraries/iDynTree/build/bindings/matlab/iDynTreeMATLAB_wrap.mexa64+00786277 mexFunction+00011605
[  2] 0x00007f67ccabbc4a        /nfsd/opt/matlab2015b/bin/glnxa64/libmex.so+00142410 mexRunMexFile+00000090
[  3] 0x00007f67ccab8244        /nfsd/opt/matlab2015b/bin/glnxa64/libmex.so+00127556
[  4] 0x00007f67ccab8de4        /nfsd/opt/matlab2015b/bin/glnxa64/libmex.so+00130532
[  5] 0x00007f67d105adbd /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_dispatcher.so+00724413 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00001549
[  6] 0x00007f67d105b250 /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_dispatcher.so+00725584 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000032
[  7] 0x00007f67cb6d58af    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+08612015
[  8] 0x00007f67cb7f97ff    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+09807871
[  9] 0x00007f67cb822196    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+09974166
[ 10] 0x00007f67cb7d79f1    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+09669105
[ 11] 0x00007f67cb41ed6c    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+05766508
[ 12] 0x00007f67cb4094c1    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+05678273
[ 13] 0x00007f67cb418075    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+05738613
[ 14] 0x00007f67cb5e9033    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+07643187
[ 15] 0x00007f67cb6c50f4    /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_lxe.so+08544500
[ 16] 0x00007f67d105adbd /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_dispatcher.so+00724413 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00001549
[ 17] 0x00007f67d105b250 /nfsd/opt/matlab2015b/bin/glnxa64/libmwm_dispatcher.so+00725584 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000032
[ 18] 0x00007f67bdb02675 /nfsd/opt/matlab2015b/bin/glnxa64/libmwmcos_impl.so+01222261
[ 19] 0x00007f67bdbc4f03 /nfsd/opt/matlab2015b/bin/glnxa64/libmwmcos_impl.so+02019075
[ 20] 0x00007f67bdb50632 /nfsd/opt/matlab2015b/bin/glnxa64/libmwmcos_impl.so+01541682
[... 168 lines cut ...]
[ 78] 0x00007f67d132b2fa      /nfsd/opt/matlab2015b/bin/glnxa64/libmwmcr.so+00594682
[ 79] 0x00007f67c4cf16b6      /nfsd/opt/matlab2015b/bin/glnxa64/libmwuix.so+00313014
[ 80] 0x00007f67c4cd7862      /nfsd/opt/matlab2015b/bin/glnxa64/libmwuix.so+00206946
[ 81] 0x00007f67d1cc10df /nfsd/opt/matlab2015b/bin/glnxa64/libmwservices.so+03002591
[ 82] 0x00007f67d1cc124c /nfsd/opt/matlab2015b/bin/glnxa64/libmwservices.so+03002956
[ 83] 0x00007f67d1cc2d9f /nfsd/opt/matlab2015b/bin/glnxa64/libmwservices.so+03009951
[ 84] 0x00007f67d1cc384c /nfsd/opt/matlab2015b/bin/glnxa64/libmwservices.so+03012684 _Z25svWS_ProcessPendingEventsiib+00000092
[ 85] 0x00007f67d132b9b8      /nfsd/opt/matlab2015b/bin/glnxa64/libmwmcr.so+00596408
[ 86] 0x00007f67d132bcd4      /nfsd/opt/matlab2015b/bin/glnxa64/libmwmcr.so+00597204
[ 87] 0x00007f67d1317fed      /nfsd/opt/matlab2015b/bin/glnxa64/libmwmcr.so+00516077
[ 88] 0x00007f67da2b9182              /lib/x86_64-linux-gnu/libpthread.so.0+00033154
[ 89] 0x00007f67d9fe647d                    /lib/x86_64-linux-gnu/libc.so.6+01025149 clone+00000109
[ 90] 0x0000000000000000                                   <unknown-module>+00000000

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

traversaro, 2015-09-30:
Mhhh so the load is crashing? I actually was able to load with no error, but then I crashed when using the object.

DiegoRomeres, 2015-09-30:
In my case load is crashing directly. I am using Matlab2015b

traversaro, 2015-09-30:
I think I know how to prevent the crash (but without a working load at the moment). I will push it in the afternoon.

DiegoRomeres, 2015-09-30:
ok, thanks! But are there any workaround to been able to load and use the data afterwards?

traversaro, 2015-09-30:
Which kind of variables? Perhaps we can write some specialized saveobj/loadobj methods for those objects that convert them to matlab objects for storing, as @naveenoid was proposing.

DiegoRomeres, 2015-09-30:
Basically I am working with the repository 'semiparametric_matlab' and the data I save executing that repository give me the error when loading them back.

At the moment the warning message is complaining with the variables belonging from the class 'IncrementalParametricEstimator'. Clearly there might be other problematic variables as well that I do not see at the moment.

traversaro, 2015-09-30:
Ok, so I should be able to replicate by running:
~~~
semiparametric_estpar_script_generalization_repeated
save datasaved.mat
load datasaved.mat 
~~~
Right?

DiegoRomeres, 2015-09-30:
I am using the file:

semiparametric_CADpar_script_generalization

but looking to the other one it should be the same.
(Anyway I will try also with semiparametric_estpar_script_generalization_repeated to see if the issue is also there)

Ommac, 2015-09-30:
Disclaimer: I have not read the whole conversation.

However, if you have to save large variables use the flag '-v7.3' of the save function, as follows:

```matlab
save(filename,'dumvar','-v7.3');
```

It happened to me that large variables were not saved properly and lead to a crash during loading.

Hope it helps.

traversaro, 2015-09-30:
Ok, i will try with
~~~
semiparametric_CADpar_script_generalization
save datasaved.mat
load datasaved.mat 
~~~

traversaro, 2016-03-16:
cc @nunoguedelha