gbionics/idyntree #78: [Estimation] Create iDynTree_Estimation library

traversaro, 2015-09-23:
As part of the estimation roadmap, we need to create an library in iDynTree related to Estimation, to collect all estimation algorithms implemented using iDynTree . 

This can be created in a `estimation` directory, and we can copy the CMake structure from another recent component, such as `high-level`. 

Code that I think should be moved or implemented in `iDynTree_Estimation` includes: 
* The [`simpleLeggedOdometry`](https://github.com/robotology/codyco-modules/blob/master/src/modules/wholeBodyDynamicsTree/src/simpleLeggedOdometry.cpp) class , that should be ported to avoid the dependency on YARP, for example using the DynamicsComputations class. 
* The [`TorqueEstimationTree`](https://github.com/robotology/idyntree/blob/master/icub/src/TorqueEstimationTree.cpp) class that should be again ported to avoid the dependency on YARP. In this case it involves more work because we need to create an iDynTree-compatible data structure to store the informations classically contained in a `dynContactList`. 
* The new class implemented by @nunoguedelha to compute the quantities required in the Berdy algorithm. For this class we will probably need to expose in the headers classes of type `Eigen::SparseMatrix`.

iron76, 2015-10-12:
What is the situation on this issue? I think we should find out who is responsible for the first point. I was thinking about either @jeljaik or @naveenoid. This is crucial for bringing forward the contact switching demo, right @DanielePucci?

naveenoid, 2015-10-13:
I had a skype call with silvio yesterday..Almost finalised the Sensors
interface. @jeljaik and I started coordinating on the design to merge with
quaternionEKF.

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Cognitive Humanoids Lab,
Department of Robotics, Brain and Cognitive Sciences (RBCS),
Istituto Italiano di Tecnologia, Genova, Italy

DanielePucci, 2015-10-13:
Yes. Without odometry a foot must be stuck on the ground.

iron76, 2015-10-13:
@DanielePucci can you explain better the last comment?

DanielePucci, 2015-10-13:
For tracking a CoM trajectory, for instance, we need its measurement w.r.t  an inertial frame. Vision along with kinematics/dynamics may be an option. For the time being, we may use a simple odemetry module that takes into account the switching of contacts, as we did for the 2nd year CoDyCo demo. Probably,  @traversaro can clarify the work we did on this if further clarifications are needed.

jeljaik, 2015-10-20:
@iron76 Regarding the first task I have finished porting this functionality from WBDT to `wholeBodyEstimator` which is still in a separate branch. The refactoring to avoid the YARP dependency remains for the moment as future work. It is however the first estimator implemented in wholeBodyEstimator.

iron76, 2015-10-20:
:+1: Can we then check one of the tasks above?

traversaro, 2016-12-10:
The discussed functionality have been introduced in the [`BerdyHelper`](https://github.com/robotology/idyntree/blob/master/src/estimation/include/iDynTree/Estimation/BerdyHelper.h) and [`ExtWrenchesAndJointTorquesEstimator`](https://github.com/robotology/idyntree/blob/master/src/estimation/include/iDynTree/Estimation/ExtWrenchesAndJointTorquesEstimator.h) classes.

closed 2016-12-10 by traversaro