gbionics/idyntree #83: Project is not open source without a license file

karan, 2015-10-19:
Without a license, this project isn't open source and no one can use the code.

Please use something like http://choosealicense.com/ to decide what license to use. I recommend MIT or GPL.

iron76, 2015-10-19:
Thanks for spotting this out. I think @traversaro will think about the proper license in the next days.

traversaro, 2015-10-19:
I will add a `LICENSE` file for iDynTree, because it is a useful convention. 

Anyway a project is perfectly open source even without a `LICENSE` file, see for example [1]:
> As long as your intention is clear, you can place your license information wherever you like. Some projects only include a descriptive line in the project's README (e.g., "This project is licensed under the terms of the MIT license."). 

@karan Given that I guess you have some sort of automatic script posting this kind of issues to random repositories to simulate a (fake) great amount of activity on GitHub [2], I would suggest you to change the automatic issue text that you post to something more correct. 

[1] https://help.github.com/articles/open-source-licensing/
[2] https://github.com/karan?tab=activity

traversaro, 2018-02-01:
Fixed by https://github.com/robotology/idyntree/pull/413 .

closed 2018-05-24 by traversaro