gbionics/idyntree #84: SWIG bindings are not regenerated when headers are modified

traversaro, 2015-10-20:
The dependency between the headers and the generated bindings should be explicitly stated in the CMake file. As a temporary workaround, a dummy modification to the `bindings/iDynTree.i` file is necessary to regenerated the bindings, as  `bindings/iDynTree.i` is correctly listed as a dependency for bindings generation.

traversaro, 2016-01-20:
In `drake` they solved this issue by using a patched `UseSWIG.cmake` file. [1][2] 
We should monitor if this works fine for them and eventually use the same solution.

[1] : https://github.com/RobotLocomotion/drake/issues/1628
[2] : https://github.com/RobotLocomotion/drake/pull/1629

naveenoid, 2016-01-20:
> [1] : RobotLocomotion/drake#1628
> [2] : RobotLocomotion/drake#1629

I find this issue to be a major pain >:( I will try to check for the drake
fix

traversaro, 2016-01-21:
To clarify: for regenerating bindings for the time being is sufficient to just run in the iDynTree source directory:
~~~
touch bindings/iDynTree.i
~~~

kjyv, 2016-01-22:
Another way to get this working is also to set SWIG_MODULE_foo_EXTRA_DEPS to include the .h files explicitly.
https://cmake.org/pipermail/cmake/2012-December/053045.html

traversaro, 2017-01-02:
This should be finally fixed in CMake 3.8 , after this MR is merged: https://gitlab.kitware.com/cmake/cmake/merge_requests/354 . It may be worth to just import the latest `UseSWIG.cmake` to benefit from this.

commit cc3d56a4 (2015-10-20): First (failed) attempt to fix #84