gbionics/idyntree #85: [idyntree] Implementation of the centroidal dynamics

DanielePucci, 2015-10-23:
For control purposes, the decoupled nature of the centroidal dynamics may be of fundamental importance in practice. So, it would be great to have all elements for running the actual control architecture with this new formalism. What it is needed is then: mass matrix,  coriolis and gravity torques, jacobians, and the term jacobians times the robot's twist (Jdot*nu).

DanielePucci, 2015-10-23:
@francesco-romano @gabrielenava

gabrielenava, 2018-09-03:
@DanielePucci concerning this very old issue, we recently discussed about the possibility of developing a piece of code that computes the derivative of the system's dynamics. I think in this case we may also think to compute the derivatives required for performing the `centroidal transformation`. [This](https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/utilityMatlabFunctions/centroidalTransformationT_TDot.m) and [this](https://github.com/robotology-playground/WBI-Toolbox-controllers/blob/master/controllers/utilityMatlabFunctions/fromFloatingToCentroidalDynamics.m) functions by default set to zero all the missing derivatives.

traversaro, 2024-03-21:
cc @diegoferigo