gbionics/idyntree #100: Gyroscope and Accelerometer sensor classes not in the `iDynTreeSensors` doxygen group

traversaro, 2015-11-23:
See http://wiki.icub.org/codyco/dox/html/idyntree/html/group__iDynTreeSensors.html ? 

@naveenoid can you take care of this issue, using the doxygen tags as explained in https://github.com/robotology/idyntree/blob/e6097038c2463a5767ba19cd9155c54bfcbeaba2/doc/dev/faqs.md#how-to-add-a-class-or-a-function-to-a-component-documentation ?

naveenoid, 2015-11-27:
https://github.com/robotology/idyntree/pull/101

traversaro, 2015-11-27:
Fixed by #101 .

closed 2015-11-27 by traversaro

traversaro, 2015-12-03:
The module tag ( `iDynTreeModelIO` ) is also missing for the sensor from urdf parser functions : http://wiki.icub.org/codyco/dox/html/idyntree/html/group__iDynTreeModelIO.html .

traversaro, 2015-12-04:
@naveenoid https://github.com/robotology/idyntree/issues/100#issuecomment-161571290

naveenoid, 2015-12-04:
Sorting out in master directly. Will keep developing my stuff in my branch