gbionics/idyntree #105: dot product is broken in bindings

traversaro, 2015-11-30:
Matlab:
~~~matlab
>> v.dot(h)
Warning: in method 'SpatialMotionVectorBase_dot', argument 2 of type 'iDynTree::DualSpace< iDynTree::SpatialMotionVector >::Type const &' 
~~~
Python:
~~~python
ERROR: testInnerProductInvariance (__main__.PositionTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "geometry.py", line 69, in testInnerProductInvariance
    power1 = f.dot(v);
  File "/home/pegua/src/codyco-superbuild/build/install/lib/python2.7/dist-packages/iDynTree.py", line 1879, in dot
    return _iDynTree.SpatialForceVectorBase_dot(self, other)
TypeError: in method 'SpatialForceVectorBase_dot', argument 2 of type 'iDynTree::DualSpace< iDynTree::SpatialForceVector >::Type const &'
~~~

Found it when re-enabling the python tests. Disabling the dot-product related tests for now. 
@nunoguedelha any idea?

traversaro, 2015-11-30:
Memo: enable the test in https://github.com/robotology/idyntree/commit/951cf53e533a2f2b9428ff85551517593a0b71de#diff-44665e26b6621156cb77e0463cfae7eaR64 when this is fixed.

nunoguedelha, 2016-08-05:
@traversaro , The issue was in `iDynTree.i`. The template `DualSpace` was not instantiated in that config file as we usually do for letting SWIG map correctly the templates. I've also added a `typedef` in the `SpatialVector.h` class declaration for getting a more readable code, no big deal. Bindings are generated and the dot product is working again. I will reactivate the dot product related tests.

nunoguedelha, 2016-08-05:
Matlab tests pass. And when running the dot product on Matlab we get no warning anymore.
Python test fails. Analysing.

nunoguedelha, 2016-08-08:
The failure was occurring only when building with Xcode and running the tests from `.../codyco-superbuild/libraries/iDynTree/build/` or `.../codyco-superbuild/build/libraries/iDynTree/`. It was fixed by https://github.com/robotology/idyntree/pull/201. I then run into another issue, this time occurring when the Python script tries to import **_iDynTree.so**:
```
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```
This is an upstream **cmake** issue related to how `find_package` command handles brewed Python. I found a work around by changing my bash profile (on MACOSX). I'm opening an new issue [here](https://github.com/robotology/idyntree/issues/202).

So now all tests, including Python tests, are passing. This issue can be closed as soon as https://github.com/robotology/idyntree/pull/201 is merged.

traversaro, 2016-08-14:
Fixed by https://github.com/robotology/idyntree/pull/201 .

closed 2016-08-14 by traversaro