gbionics/idyntree #108: Doxygen documentation for SpatialMotionVector and SpatialForceVector is difficult to understand

traversaro, 2015-12-03:
See http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1SpatialMotionVector.html . 

In particular the typedef's  `LinearVector3T` and `AngularVector3T` are hiding the types actually accepted/returned by the methods.

Do you think we can do anything for handling this @nunoguedelha ?

traversaro, 2015-12-03:
@nunoguedelha   Perhaps we overload the `get**Vec3`  function in the `Spatial***Vector` classes to make sure the right type appears in the doxygen documentation?

nunoguedelha, 2015-12-04:
as you must have understood, the `LinearVector3T` and `AngularVector3T` are mapped to the 3D vector types associated to `SpatialMotionVector` or `SpatialForceVector`. So they are compiled as types `LinearMotionVector3`, `AngularMotionVector3`, `LinearForceVector3` or `AngularForceVector3`, which are the actual types returned by the methods.

But I just realised that the keyword `typename` was missing from the typedef definitions of `LinearVector3T` and `AngularVector3T` in `SpatialMotionVector` and `SpatialForceVector` classes. That should be causing the problem, because then the compiler doesn't know that `SpatialMotionForceVectorT_traits<SpatialForceVector>::LinearVector3Type` is a type.

nunoguedelha, 2015-12-04:
And by the way, I think we don't even need the `typedef` definitions in `Spatial**Vector` since they should be inherited by the base class SpatialVector. I can improve the documentation in these files.

traversaro, 2015-12-04:
Yes, I have no problem in understanding the CRTP structure of the classes, my concern was more on users having to use those classes (consider that we automatically produce that documentation with [doxygen](www.doxygen.org) , that does not include a full compiler inside it.

nunoguedelha, 2015-12-04:
Yes, I see, indeed, the user needs some clicks and looking at the file `PrivateMotionForceVertorAssociations.h` to get the real type. The problem is that what triggered the use of CRTP here is actually the intention to avoid overloading methods on derived classes (for the only purpose of defining the expected return types). Because then you start overloading all those methods, which is way less efficient for the idyntree-core developpers, and the users. In my opinion, it's better to add comments, or a solution to "resolve" typedefs in documentation, instead of dropping the advantage of CRTP. We will potentially have the same issue every time we use templates and CRTPs.

nunoguedelha, 2015-12-04:
I think we could find some solution out there for resolving typedefs in documentation. It's worth it and a general improvement for documentation.

traversaro, 2015-12-04:
> The problem is that what triggered the use of CRTP here is actually the intention to avoid overloading methods on derived classes (for the only purpose of defining the expected return types). [...]

I agree in general, if is possible to make the class understandable for users of the library. 
Why do you think that overloading those methods is less efficient? Once you defined them for `SpatialMotionVector` and for `SpatialForceVector` we are done, in which sense they are not efficient?

traversaro, 2015-12-04:
(By the way I opened this issue given by the real world feedback that I received from @naveenoid that needed to use that class).

nunoguedelha, 2015-12-04:
I wasn't only talking about `SpatialMotionVector` and `SpatialForceVector`, but all the classes using CRTP or templates. For instance `MotionVector3` (http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1MotionVector3.html). The return type `MotionCrossLinM` has the same documentation problem.
What I mean is that this is a general problem with automatic documentation and templates/CRTPs. So either we drop templates (I wouldn't do that for the sake of automatic documentation), either we add comments. Addressing `SpatialMotionVector` and `SpatialForceVector` is just a particular use case...

nunoguedelha, 2015-12-04:
ok, connecting again to the real world :) . If it really helps overloading at least for the most used classes, then do it, but I still think we loose the interest of CRTP, or part of it. If I'm analysing a code, I prefer to clearly see which methods are really being changed from a Base class to a derived one. So I would go for adding comments for the users.

traversaro, 2015-12-04:
Take your time to think about a solution that seems good to you. 
Consider anyway that the time spent by everyone in the group (and by partners in EU projects) in reading those base classes and their documentation  will probably be one or two order of magnitude bigger than the time we spent implementing them. For this reason I think that moving the complexity burden from the user reading the class/the documentation to the programmer that implemented it is a worthy tradeoff.

nunoguedelha, 2015-12-04:
Yes Silvio, fair enough. We can overload the methods for `SpatialMotionVector` and `SpatialForceVector` if these are the most visible, and later revert if we find a better solution. Let's keep an eye on possible solutions for a better documentation of template classes.