gbionics/idyntree #111: iDynTree does not work with CMake 2.8.11

traversaro, 2015-12-10:
[image]

traversaro, 2015-12-10:
The problem is in the use of the keyword `SYSTEM` in the `target_include_directories` command, that was introduced in CMake 2.8.12 .
 While we should support CMake 2.8.11 , most of the users and the travis CI actually test it against 2.8.12, and that's the reason why this got unnoticed. 
Removing the `SYSTEM` keyword for now, given that is not essential. 

cc @francesco-romano @DanielePucci

closed 2015-12-10 by traversaro

traversaro, 2016-03-17:
@DanielePucci and @pi-q found another incompatibily in the code, fixed by https://github.com/robotology/idyntree/commit/dfd109594b448296101e6c04f2545ab89d93cf5b . 
In particular PUBLIC and PRIVATE options of target_link_libraries are not
supported in CMake 2.8.11 [1] (they have been introduced in 2.8.12 [2]), LINK_PRIVATE and LINK_PUBLIC should be used instead.

[1] : https://cmake.org/cmake/help/v2.8.11/cmake.html#command:target_link_libraries
[2] : https://cmake.org/cmake/help/v2.8.12/cmake.html#command:target_link_libraries

francesco-romano, 2016-08-24:
I haven't taken a look at the `CMakeLists` files, but `SYSTEM` is a useful feature and we should support is where possible.

Regarding the tests with travis (or other CIs), we could think of something like the following.
We can create a repository where we can host versioned brew formula of the packages (like cmake 2.8.11), tap that repo and install that particular formula.
We can take the formula from the history of home-brew github repository, e.g. https://github.com/Homebrew/homebrew-core/blob/8f0fdd4c4fe5e969dbcdebb360927a6610e6e2cb/Formula/cmake.rb

traversaro, 2016-08-24:
> I haven't taken a look at the CMakeLists files, but SYSTEM is a useful feature and we should support is where possible. 

My current approach is to substitute: `target_include_directories(<target> SYSTEM <include_dirs>)` with `include_directories(SYSTEM <include_dirs>)`, because the latter is available on CMake 2.8.11 .

commit b370b7fd (2015-12-10): Fix #111 - Incompatibility with CMake 2.8.11