gbionics/idyntree #113: Google still points to the old doxygen documentation

traversaro, 2015-12-11:
Some time ago I moved the iDynTree documentation from being generated with `codyco-superbuild` doxygen (at http://wiki.icub.org/codyco/dox/html/index.html) to its own doxygen website at http://wiki.icub.org/codyco/dox/html/idyntree/html/ , for cleaning up its the docs. 

Unfortunately Google still points to the old documentation for a lot of pages. If you search "iDynTree Transform" on Google, for example, the first result that I get is : 
~~~
http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1Transform.html
~~~
instead of the correct http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1Transform.html (I avoided to link the old version to avoid boosting its relevancy in Google results). 

This is a **big** problem, because the google search is quite a natural entry point for users (I get the 404 error several times a day, and is annoying even if I know the underlying issue). 

We need to solve this. We cannot put an automatic redirect or a HTTP 301 on `http://wiki.icub.org/codyco/dox/html/`, because that prefix is still used by the rest of the CoDyCo documentation. 
Possible solutions involve:
 * Remove all existing links to the old documentation from the internet (probably there are some around), to reduce its pagerank. 
 * Boost the pagerank of the new doxygen documentation adding links to it around in icub/codyco websites. 
 * Provide a sitemap for both codyco and idyntree doxygen documentation, so that Google can know exactly which pages are available in each doxygen website.

kjyv, 2016-01-08:
It seems both link examples you give, old and new are the same. The google results load fine for me now and are at the new location.

traversaro, 2016-01-08:
Yes, it seems that Google has update its indeces and for most users the new doxygen repo result is higher then the old repo. I think we can consider this issue closed by Google.

closed 2016-01-08 by traversaro