gbionics/idyntree #114: Use mscgen to improve documentation

traversaro, 2015-12-11:
[`mscgen`](http://www.mcternan.me.uk/mscgen/) is a nice tool to produce Message Sequence Chart diagrams, that can be useful to document complicate function structure. 

It can be also useful to document interaction between threads/process/middleware components, but that is out of scope for iDynTree because we only keep C++ code that address the concern of **computation** out of the [5C concerns](http://people.mech.kuleuven.be/~bruyninc/ecs/LevelsOfComplexity-5C-20110223.pdf) . 

Luckily doxygen support `mscgen` out of the box, so we can easily embed msc diagrams in existing doxygen comments. 

The step related to this are then: 
 * [ ] Add support for finding and using `mscgen` in the Doxyfile produced by CMake (@traversaro) 
 * [ ] Document a function/class with a msc, to have a first example on its use (@nunoguedelha)
 * [ ] Install `mscgen` on the machine that produces the doxygen documentation that is published at http://wiki.icub.org/codyco/dox/html/idyntree/html/ . It should not be difficult, given that `mscgen` is available in stock Debian repositories.

nunoguedelha, 2015-12-11:
Also, `GraphViz` is the open-source tool used by `Doxygen` for generating diagrams of class hierarchy and collaboration. It generates graphs specified in DOT language scripts. We can use this for generating additional function call or class relations diagrams which have not been automatically generated by Oxygen, or diagrams related to classes and methods yet to implement.
I will use it along with `mscgen` in my next documentation.

traversaro, 2015-12-11:
Which doxygen command do you plan to use in particular?

nunoguedelha, 2015-12-12:
Doxygen can be configured to use the `dot` tool from GraphViz to generate diagrams more advanced than the inheritance diagrams generated through the default built-in tool. The `dot` tool has to be in the path and `HAVE_DOT` has to be set in the Doxygen configuration file. There's a lot of information in the links below (that you might already know):
https://www.stack.nl/~dimitri/doxygen/manual/index.html
https://www.stack.nl/~dimitri/doxygen/manual/diagrams.html
https://www.stack.nl/~dimitri/doxygen/manual/config.html#cfg_have_dot

There might also be a way to include additional dot diagrams in Doxygen documentation without impacting the way inheritance diagrams are generated. I have to read more on that.

traversaro, 2015-12-12:
Perfect, then the cmake code for setting the `DOT_PATH` variable in OS in which the installed dot is not in the path are the same done in https://github.com/robotology/idyntree/pull/115 for mscgen .

nunoguedelha, 2015-12-13:
@traversaro I have actually tried out the `HAVE_DOT = YES`, it improves a lot the documentation of the class hierarchy as well as the illustration of template class instantiation. Here's the flag configuration set in `Doxyfile.in`:

Tag | Value | Description
------|---------|----------------
GENERATE_TREEVIEW | YES | generate a tree-like index structure to display hierarchical information
HAVE_DOT | YES | use dot tool to generate graphs
TEMPLATE_RELATIONS | YES | inheritance and collaboration graphs show the relations between templates and their instances
CALL_GRAPH | YES | generate a call dependency graph for every global function or class method
CALLER_GRAPH | YES | generate a caller dependency graph for every global function or class method

If necessary, in order to time when generating the documentation, we can restrict this feature to selected functions by using the `\callgraph` command in the function documentation. Same for `caller_graph` (`\callergraph`). I didn't use it, but `INTERACTIVE_SVG` allows you to zoom and pan when you choose to generate `svg` images instead of `png`.

nunoguedelha, 2015-12-13:
screenshot of the result for `Transform` class:
[image]

traversaro, 2015-12-13:
:+1: , Can you open a PR with those options added in the `Doxyfile` ? 
I then want to make sure that dot and mscgen are installed on the machine that generates the doxygen by asking to Matteo Brunettini.

nunoguedelha, 2015-12-13:
yes, will do ASAP.

nunoguedelha, 2015-12-14:
@traversaro done

traversaro, 2015-12-14:
`mscgen` installed on the doxygen machine by @mbrunettini .

traversaro, 2016-06-27:
See https://github.com/robotology/idyntree/pull/174 .

traversaro, 2019-10-07:
I think in the end we never used mscgen, so I think we can close this issue, what do you think @nunoguedelha ?

nunoguedelha, 2019-10-07:
I see that I never did step 2 of the task list..
Do you think it's not worth to push people to document functions that they change or implement in the future? I don't have a strong opinion on this..

traversaro, 2019-10-07:
> Do you think it's not worth to push people to document functions that they change or implement in the future? I don't have a strong opinion on this..

I typically require doxygen docs for the function, but the need of sequence diagrams never came out as a requirement. If someone needs mscgen in the future, he/she will be able to restore this code easily.

nunoguedelha, 2019-10-07:
Yes, sure, we can close it.

closed 2019-10-07 by traversaro