gbionics/idyntree #126: segfaults when using DynamicsRegressorGenerator from python

kjyv, 2016-01-08:
Trying the DynamicsRegressorGenerator class from Python, I get segfaults when calling getNrOfDegreesOfFreedom() and setRobotState() respectively. Not sure if it is me doing something wrong though. The same URDF file works fine with the DynamicsComputations class and other URDF files show the same behavior.
See this gist for an example: https://gist.github.com/kjyv/7d3eaf99420c595b7a6b

traversaro, 2016-01-08:
Hi @kjyv , 
 at the moment the `DynamicsRegressorGenerator` support several different regressors (and multiple combination of the them), so it needs to load the regressors structure from a XML string (see this matlab example: https://github.com/robotology/idyntree/blob/master/examples/matlab/iCubLeftArmDynamicRegressors/iCubLeftArmFTRegressor.m ).

Clearly the fact that the python interpreters segfaults instead of printing a nice error message is a bug, and I plan to fix it in the weekend. 

If you want to compute the identifiable space using for the complete dynamics regressors (the one returned by [DynamicsComputations::getDynamicsRegressor](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1HighLevel_1_1DynamicsComputations.html#a37f3c10c02a2657adeccefddcaac1685) we will need this "complete" dynamics regressor to the class of regressors supported by the `DynamicsRegressorGenerator` class. This should be doable because the code for computing it is already implemented, and I just need to add the proper XML parsing to add this regressor. If you want to discuss your needs in persons (and your at IIT) you can pass by my desk, I am in iCub Facility near to the red iCub.

kjyv, 2016-01-11:
Adding a regressor structure and adding an FT sensor to the URDF file now results in some results.

closed 2016-01-11 by kjyv