gbionics/idyntree #135: Support std::vector<std::string> function parameters in Matlab/Python

traversaro, 2016-02-09:
We started to use a lot of instances of `std::vector<std::string>` as input/output parameter of function and methods. We need to properly modify SWIG `.i` configuration files to make sure that this function can be used in Matlab. 
The first step (currently investigated by @nunoguedelha ) is to just instantiate the `std::vector<std::string>` template during the SWIG bindings generation.

A cleaner way would be to define proper typemaps between `std::vector<std::string>` and the appropriate data structure in the target language (cell array of strings in matlab, list of strings in python, table of strings in lua).

naveenoid, 2016-02-10:
:+1:

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Dynamic Interaction Control Lab,
iCub Facility,
Istituto Italiano di Tecnologia, Genova, Italy

traversaro, 2016-07-21:
The `StringVector` template instantiation can be easily populated with the `push_back` method, but at least in Matlab I did not find any way of retrieving its contents. I wonder if it something due that should be solved upstream in `https://github.com/jaeandersson/swig/blob/matlab/Lib/matlab/std_vector.i` or what.

prashanthr05, 2019-06-10:
Any knowledge update regarding handling `std::vector<T>` of generic objects? Thanks in advance!

traversaro, 2019-06-10:
Like an automatic typemap that converts `std::vector<iDynTree::ComplexObject>` to a cell array of `iDynTree.ComplexObject` Matlab objects? To be honest I do not know how to handle that.

prashanthr05, 2019-06-10:
I was trying to handle a vector of rotations like `std::vector<iDynTree::Rotation>`,  so I created a template in `iDynTree.i` as `%template(RotationVector) std::vector<iDynTree::Rotation>;`
and this enables the use of standard `std::vector` methods to handle `iDynTree::Rotations`

traversaro, 2019-06-10:
Ok, that seems to be a solution in line with the `StringVector` strategy.