gbionics/idyntree #144: CoM f.k. and jacobian

francesco-romano, 2016-04-04:
plus transformation from angular velocity to derivative of quaternion.

francesco-romano, 2016-04-05:
and `getRelativeFrameJacobian`

traversaro, 2016-04-07:
First COM-related methods added in https://github.com/robotology/idyntree/pull/146 .

francesco-romano, 2016-05-27:
Adding other useful methods:

- Rotation::asQuaternion
- Sparsity pattern for Jacobian and Hessians

traversaro, 2016-05-27:
What would be a good format for sparsity patterns (i.e. what is used by IPOPT?).

francesco-romano, 2016-05-27:
From what I understood from the documentation the format is quite free, in the sense that the Jacobian is specified as a one-d vector.
Then the correspondence between position in the vector and in the original matrix is supplied by the user by filling two vector.
This example fills a dense matrix of 2 x 4:
```
// this particular Jacobian is dense
    iRow[0] = 0; jCol[0] = 0;
    iRow[1] = 0; jCol[1] = 1;
    iRow[2] = 0; jCol[2] = 2;
    iRow[3] = 0; jCol[3] = 3;
    iRow[4] = 1; jCol[4] = 0;
    iRow[5] = 1; jCol[5] = 1;
    iRow[6] = 1; jCol[6] = 2;
    iRow[7] = 1; jCol[7] = 3;
```
As you can see we are saying that the 8th position in the vector is the element (2,4) (one based in my 'math' description 😄 )

traversaro, 2016-05-27:
Mhh, this seems a classical sparse matrix representation. Just for the sake of understanding, does IPOPT support both sparse (this one) matrices and dense matrices rappresentation or just this sparse one?

traversaro, 2016-05-27:
If we are constrained to the sparse representation, I would go for writing a basic `iDynTree::MatrixSparse` class compatible with both IPOPT and [Eigen::Map<SparseMatrix>](https://eigen.tuxfamily.org/dox-devel/classEigen_1_1Map_3_01SparseMatrixType_01_4.html), and then let Eigen handle all the dirty job.

francesco-romano, 2016-05-28:
I've no experience on sparse matrix representation, so I let you decide the format.

IPOpt support only this format. Thus you can fill a "dense-sparse" matrix :)

francesco-romano, 2016-05-31:
First quaternion conversion implemented here: https://github.com/robotology/idyntree/pull/167

traversaro, 2018-02-07:
I guess this issue was fixed in the end, feel free to open again otherwise.

closed 2018-02-07 by traversaro