gbionics/idyntree #153: Full Test Suite does not run on OS X

traversaro, 2016-04-19:
* [ ] Valgrind fails because of some false positives on OS X standard library. 
* [ ] Python test fails ( https://travis-ci.org/robotology/idyntree/jobs/124131468#L6639 ):
~~~
44/44 Test #44: PythonGeometryTests ..............................***Exception: Other  1.54 sec

Fatal Python error: PyThreadState_Get: no current thread
~~~

kjyv, 2016-04-19:
I got that too, apparently it has to do with using different library while building vs. interpreter when running.

traversaro, 2016-04-19:
I see.. I am 
just delaying all the fixes that are not immediate to finally get travis back to green.

traversaro, 2019-08-29:
Similar error in GitHub Actions: https://github.com/robotology/idyntree/pull/568/checks?check_run_id=206937776 / related commits: e03f88a . .