gbionics/idyntree #159: Error using iDynTree with Python bindings

gtournois, 2016-05-17:
Hi! 

At the IIT I have been working with iDynTree-superbuild using the Python bindings for parameter estimation of HyQ. Recently I have installed all necessities on another pc, but now I am not able to run my program.

The cause being, some of the matrices I obtain from iDynTree contain nan's. After some research, I think this is due to incorrect or the absence of initialization of one of the iDynTree matrices. However, I don't know which one.

How would I be able to fix this? What has changed since September last year that can cause this?

Thanks!

traversaro, 2016-05-17:
Yes, there was a round of optimization (documented in https://github.com/robotology/idyntree/issues/98) in which we removed the initialization of classes that are not dynamically allocated. 

The change is documented in: 
* https://github.com/robotology/idyntree/blob/master/MIGRATING.md#idyntree-03x-to-04x 
* https://github.com/robotology/idyntree/issues/128 .

I just noticed that I did not update the python example that you were using as a reference, but I just did that in https://github.com/robotology/idyntree/commit/a9d5b8ab5eadb07b027ebafd996fd3b0c8ae67a3 . Let me know if this solves your problems!

gtournois, 2016-05-17:
Thanks so much! This solved everything.

traversaro, 2016-05-17:
Perfect!

closed 2016-05-17 by traversaro