gbionics/idyntree #162: Wrong IMU proper acceleration estimations

nunoguedelha, 2016-05-19:
We have been observing several inconsistencies in the inertial sensors measurement predictions (computed in `computeExpectedFTSensorsMeasurements()`). The results are pretty clear for the IMU sensor (refer to attached plots).
@traversaro The cause seems to be in the [`AccelerometerSensor::predictMeasurement(...)`](https://github.com/robotology/idyntree/blob/9e62ff6b886003d697a4b5256fad0e3080f26ce5/src/sensors/src/AccelerometerSensor.cpp#L155) method, where the link acceleration is being projected to the sensor frame using `this->pimpl->link_H_sensor` when it should use the inverse transform. Same observation for the computation of the local velocity.
`this->pimpl->link_H_sensor` is actually described in the class `AccelerometerSensor ` as being the transform from link to sensor, which makes the notation "link_H_sensor" confusing. Anyway, I could observe, for a specific joint configuration (qi = 0), that `link_H_sensor` matches the expected transform projecting an IMU measurement into the link frame (the "head" in this case). We should have, considering a null local velocity :
```
acc_{imuFrame} = link_H_sensor.inverse() * acc_{link}
```
To be tested.

traversaro, 2016-05-19:
Great catch!
 There should be a test covering this, so probably also the test is bugged, looking at it.

nunoguedelha, 2016-06-21:
@traversaro  Added the figure showing the issue. Tested the fix. Pushing it now.
[image]

nunoguedelha, 2016-06-21:
Fixed in https://github.com/robotology/idyntree/pull/171.

closed 2016-06-23 by traversaro