gbionics/idyntree #163: Linker error par2urdf: undefined reference to urdf::parseURDF(std::string const&)

rlober, 2016-05-19:
With all of superbuild up to date I get the following linker error using gcc/g++ 4.8.5:
```
Scanning dependencies of target idyntree-icub-kdl
[ 93%] Building CXX object src/icub-kdl/CMakeFiles/idyntree-icub-kdl.dir/src/TorqueEstimationTree.cpp.o
[ 94%] Building CXX object src/icub-kdl/CMakeFiles/idyntree-icub-kdl.dir/src/iDyn2KDL.cpp.o
[ 95%] Building CXX object src/icub-kdl/CMakeFiles/idyntree-icub-kdl.dir/src/idyn2kdl_icub.cpp.o
[ 95%] Linking CXX shared library libidyntree-icub-kdl.so
[ 95%] Built target idyntree-icub-kdl
[ 96%] Built target idyntree-high-level-kdl
[ 96%] Linking CXX executable par2urdf
../model_io/urdf-kdl/libidyntree-modelio-urdf-kdl.so: undefined reference to `urdf::parseURDF(std::string const&)'
collect2: error: ld returned 1 exit status
src/tools/CMakeFiles/par2urdf.dir/build.make:110: recipe for target 'src/tools/par2urdf' failed
make[5]: *** [src/tools/par2urdf] Error 1
CMakeFiles/Makefile2:1261: recipe for target 'src/tools/CMakeFiles/par2urdf.dir/all' failed
make[4]: *** [src/tools/CMakeFiles/par2urdf.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/iDynTree.dir/build.make:107: recipe for target 'libraries/iDynTree/CMakeFiles/YCMStamp/iDynTree-build' failed
make[2]: *** [libraries/iDynTree/CMakeFiles/YCMStamp/iDynTree-build] Error 2
CMakeFiles/Makefile2:1024: recipe for target 'CMakeFiles/iDynTree.dir/all' failed
make[1]: *** [CMakeFiles/iDynTree.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```

traversaro, 2016-05-19:
On which os?

rlober, 2016-05-19:
Ubuntu ... 16.4 :(

traversaro, 2016-05-19:
I am just compiling on it, let's check!

rlober, 2016-05-19:
Cooooooool

traversaro, 2016-05-19:
It is working fine for me, but my gcc is 5.3 : 
~~~
 gcc --version
gcc (Ubuntu 5.3.1-14ubuntu2) 5.3.1 20160413
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
~~~
Any reason on why you want you use gcc 4.8 on 16.04 ? How do I use that one?

rlober, 2016-05-19:
I want to use gcc4.8 because we have some unexpected behavior in ocra-recipes and we are trying to figure out the source of the problem. To change it just do:
```
export CC="gcc-4.8"
export CXX="g++-4.8"
```
When you run cmake for the first time in superbuild it should tell you if it is using 5... or 4...

traversaro, 2016-05-19:
I was able to replicate a similar error, and the linkin error was due to the fact that I had shared libraries and binaries compiled with gcc5 that tried to link shared libraries compiled with `gcc4`. The fishy point was that all the error were reporting linking errors on function that had `std::string` in the signature, that is one of the classes that changed ABI between gcc4 and gcc5.

rlober, 2016-05-20:
@traversaro seems to have resolved this problem by not using the system's tinyxml and forcing compilation with gcc4.8. I will let him fill in any details but it seems like this can be closed.

closed 2016-05-20 by rlober