gbionics/idyntree #175: Fix idiosincracity in RevoluteJoint constructor

traversaro, 2016-06-27:


aikolina, 2018-03-12:
I have no idea of idiosincracity's meaning, but does this issue mean that the `RevoluteJoint` constructor might not work correctly?

If this is the case, I have a problem to use this constructor in Matlab.

I have tried [all the possible variants](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1RevoluteJoint.html#aa09aee905d118d03a9d22b5bdc98e132) (two/four/no arguments), and I always get the error message **No matching constructor**.

```Matlab
>> planar_model
Error using iDynTree.RevoluteJoint (line 85)
No matching constructor

Error in planar_model (line 37)
ankleJoint = iDynTree.RevoluteJoint();
 
>> planar_model
Error using iDynTree.RevoluteJoint (line 85)
No matching constructor

Error in planar_model (line 38)
ankleJoint = iDynTree.RevoluteJoint(baseLink, lowerLegLink, baseLink_H_lowerLegLink, rotationAxis);
 
>> planar_model
Error using iDynTree.RevoluteJoint (line 85)
No matching constructor

Error in planar_model (line 39)
ankleJoint = iDynTree.RevoluteJoint(baseLink_H_lowerLegLink, rotationAxis);

```

traversaro, 2018-03-13:
The idiosyncraticy is something else, moved the issue with Matlab joint constructors in https://github.com/robotology/idyntree/issues/424 .

francesco-romano, 2018-03-13:
@traversaro is trying to solve the always spreading ignorance 😂

traversaro, 2018-03-13:
:D