gbionics/idyntree #184: generate python 3 compatbile bindings

kjyv, 2016-07-14:
Trying to `import iDynTree` from python 3 will try to import the python2.7 version which however fails due to non-compatible code. The swig help suggests that the generated code should be python 3 compatible automatically (http://www.swig.org/Doc3.0/SWIGDocumentation.html#Python_python3support) but that seems to not be directly working or I'm missing something.

traversaro, 2016-07-14:
I think you need to specify `IDYNTREE_USES_PYTHON_VERSION ` (see https://github.com/robotology/idyntree/blob/master/bindings/python/CMakeLists.txt#L2) CMake variable to select for which python you want to compile the bindings (if you have multiple versions installed, otherwise CMake should automatically found the only one present in the system). 

Building the bindings for both Python 2.7 and 3 I guess it involves building the two different bindings and copy them in different directories, so it would probably involve some changes in the CMakeLists.txt files.

kjyv, 2016-07-14:
Aha, ok! But the way I understand the SWIG docs are that it should already build it for both.

traversaro, 2016-07-14:
The generated C++/Python code is compatible (at API level) with both Python 2 and 3, but then when you compile you need to use the include and link a precise version of Python libraries (because they are not ABI compatible).

kjyv, 2016-07-14:
Ok, so it would be a nice to have to build both but for now it is choosable, good.

traversaro, 2016-07-14:
Yes, I guess they probably need to go in different installation prefix but I did not check the issue in depth.