gbionics/idyntree #194: SimpleLeggedOdometry drifts on z-axis

traversaro, 2016-07-25:
cc @hu-yue 

Z-axis plot of the output of SimpleLeggedOdometry  : 
[image]

hu-yue, 2016-07-25:
Actually there is drift also on the other axis.
I tested a long walk (about 20 steps) and in the end the difference between "reference" and estimation is:
* On X-axis: 0.05 [m]
* On Y-axis: 0.02 [m]
* On Z-axis: 0.015 [m]
Plot:
[image]

hu-yue, 2016-07-26:
I also noticed that the yaw rotation (3rd element of base:o) is sometimes switched from -180 to 180 for few samples:
[image]

traversaro, 2016-12-10:
This was actually due to the mismatch highlighted in https://github.com/robotology-playground/icub-model-generator/pull/15#issue-170126390 .