gbionics/idyntree #203: Indexes issue using Matlab bindings

claudia-lat, 2016-08-09:
I observed the following issue in using the Matlab bindings: it can happen that simultaneously in your code you have zero- and one-based indexes.  I'm not discussing about of what is the best choice but   of the confusion in having both of them.  Is this issue improvable? 

Another thing I noticed: having the (wrong) habit to open the Matlab variables from the workspace, when you deal with iDynTree object and keep open a workspace variable  during the workflow it is not automatically updated; if you want to see its updated value  you have to close and re-open it.  Have you ever noticed this?

traversaro, 2016-08-14:
> it can happen that simultaneously in your code you have zero- and one-based indexes. I'm not discussing about of what is the best choice but of the confusion in having both of them. Is this issue improvable? 

I think that making sure that `iDynTree` classes are *always* 0-based while usual matlab stuff if 1-based is the best that we can do on this. Adding typemaps to avoid the overhead of converting back and forth matlab quantities to iDynTree quantities could partially resolve this, reducing the number of  iDynTree-specific quantities used in Matlab user code. 

> Another thing I noticed: having the (wrong) habit to open the Matlab variables from the workspace, when you deal with iDynTree object and keep open a workspace variable during the workflow it is not [...]

What do you mean by "open the Matlab variables from the workspace"?

francesco-romano, 2016-08-15:
Variables can be "double-clicked" on the workspace and inspected in a separate window.
That window is synched with the actual values of the variables, while iDynTree variables not (it seems as if a toString function is called at the time of window opening)

[image]