gbionics/idyntree #205: Function for selecting sensors loaded from URDF

claudia-lat, 2016-08-12:
Consider the following use case: in your URDF model you have a list of sensors that you load by using `ModelLoader` but you want  to use only some of these sensors in the computation.  Actually it doesn't exist a function that allows you to select the sensors. It can be useful to add it.

traversaro, 2016-08-14:
I think in most cases you will have a list of "selected" sensors that you want loaded, and you then discards all the sensors that are not part of this "selected" sensors, similarly to what we already do for joints in the [`loadReducedModelFromFullModel`](http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1ModelLoader.html#aa1d105cc4c13bc5c6fd90cc6f8224085) method.

francesco-romano, 2016-08-15:
@traversaro we can then add an additional parameter to `loadReducedModelFromFullModel` with the list of available sensors.
Anyway I added some `remove` functions to the `SensorsList` class

PR #211 (2016-08-19): Adding functionalities for Berdy
Related issues:
Fix https://github.com/robotology/idyntree/issues/204
Fix #205 

Still missing:

- [x] update `loadReducedModel*` taking into account the list of sensors
- [x] provide dynamic variables ordering (as we provide sensors ordering)

cc @traversaro @claudia-lat
closed 2016-08-19 by francesco-romano