gbionics/idyntree #209: Names of `Joint`/`Link` classes

traversaro, 2016-08-16:
Currently we are storing names of joint and links in the Model, rather then in the Joint/Link classes itself. When I originally wrote those classes I was quite concerned about the possible hit on performance due to the cache misses induced by having a large string in the `Link` vector, for example (especially considering that the new C++11 standard enforces small string optimization in `std::string`). 

However now we have a proper benchmarks in https://github.com/robotology/idyntree/blob/master/src/tests/benchmark/DynamicsBenchmark.cpp , so we can re-evaluate that decision.

traversaro, 2016-08-16:
cc @francesco-romano

francesco-romano, 2016-08-16:
👍