gbionics/idyntree #214: Add xsd of urdf format supported by iDynTree

traversaro, 2016-08-18:
iDynTree supports the urdf format informally defined in http://wiki.ros.org/urdf/XML plus some extensions (defined in https://github.com/robotology/idyntree/blob/master/doc/model_loading.md).

It would be convenient to add an xsd of the urdf format supported by iDynTree, based on the upstream one available at  https://github.com/ros/urdfdom/blob/master/xsd/urdf.xsd .

francesco-romano, 2016-08-29:
If we want to add support to validation, I fear (but I'm happy) to move away from `tinyxml`.
A good alternative can be [Xerces](http://xerces.apache.org/xerces-c/program-3.html)

francesco-romano, 2016-08-31:
xerces-c is available on brew (3.1.4), ubuntu and debian (3.1.1 which are end of 2014, 3.1.3 on new ubuntu and debian testing)