gbionics/idyntree #221: Visualiser does not work in interpreted languages

francesco-romano, 2016-08-25:
e.g. when executing the following code in MATLAB
```matlab
viz = iDynTree.Visualizer();
viz.init();

```

this error appears in the terminal (not matlab prompt)
```
Irrlicht Engine version 1.8.3
2016-08-25 11:39:17.819 MATLAB_maci64[25749:21542325] _createMenuRef called with existing principal MenuRef already associated with menu
2016-08-25 11:39:17.824 MATLAB_maci64[25749:21542325] (
	0   CoreFoundation                      0x00007fff92f7b4f2 __exceptionPreprocess + 178
	1   libobjc.A.dylib                     0x00007fff8b5f8f7e objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff92fe24bd +[NSException raise:format:] + 205
	3   AppKit                              0x00007fff9584cc86 -[NSCarbonMenuImpl _createMenuRef] + 62
	4   AppKit                              0x00007fff9584c5c5 -[NSCarbonMenuImpl _instantiateCarbonMenu] + 140
	5   AppKit                              0x00007fff9584a270 -[NSApplication finishLaunching] + 856
	6   libidyntree-visualization.dylib     0x000000017270cac7 _ZN3irr16CIrrDeviceMacOSXC1ERKNS_27SIrrlichtCreationParametersE + 503
	7   libidyntree-visualization.dylib     0x000000017269fb61 createDeviceEx + 49
	8   libidyntree-visualization.dylib     0x000000017250d3b1 _ZN8iDynTree10Visualizer4initENS_17VisualizerOptionsE + 273
	9   iDynTreeMEX.mexmaci64               0x0000000171de6c1e _Z29_wrap_Visualizer_init__SWIG_1iPP11mxArray_tagiS1_ + 78
	10  iDynTreeMEX.mexmaci64               0x0000000171de6cd0 _Z21_wrap_Visualizer_initiPP11mxArray_tagiS1_ + 80
	11  iDynTreeMEX.mexmaci64               0x0000000171df15b1 mexFunction + 12049
	12  libmex.dylib                        0x0000000111bbfd34 mexRunMexFile + 84
	13  libmex.dylib                        0x0000000111bbc0d1 _ZN7Mfh_mex30runMexFileWithSignalProtectionEiPP11mxArray_tagiS2_ + 113
	14  libmex.dylib                        0x0000000111bbc25b _ZN7Mfh_mex13dispatch_fileEiPP11mxArray_tagiS2_ + 203
	15  libmwm_dispatcher.dylib             0x000000011277c47c _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_ + 796
	16  libmwm_dispatcher.dylib             0x000000011277c8f3 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_ + 35
	17  libmwm_lxe.dylib                    0x0000000113ab1afe _ZN9MathWorks3lxe17LxeDispatcherImpl12apply_handleEPNS0_29DispatcherFunctionHandleTokenEmPNS0_6xvalueEmPKS4_ + 910
	18  libmwm_lxe.dylib                    0x0000000113c8270a _ZN9MathWorks3lxe18ApplyNargoutCommonIXadL_ZNS0_17ApplyHandleCommonEPvNS_2ts4TypeEPKNS0_6xvalueES2_S2_mmEEEEvS2_S4_PKvS2_S2_ + 474
	19  libmwm_lxe.dylib                    0x0000000113c00d05 _ZN9MathWorks3lxe38cpp_primitive_template_fn_with_contextIXadL_ZNS0_20apply_handle_nargoutEPvNS_2ts4TypeEPKvS2_S2_EEEEbS2_S4_S6_S2_S2_ + 21
	20  libmwm_lxe.dylib                    0x000000011363b177 _ZN9MathWorks3lxe13IrInterpreter17ApplyCppPrimitiveEPN2ir4termES4_RKNS_2ts4TypeES8_ + 407
	21  libmwm_lxe.dylib                    0x000000011363a36f _ZN9MathWorks3lxe13IrInterpreter31process_application_of_constantEPN2ir4termE + 1999
	22  libmwm_lxe.dylib                    0x0000000113638838 _ZN9MathWorks3lxe13IrInterpreter13applyConstantEPN2ir4termE + 248
	23  libmwm_lxe.dylib                    0x0000000113636d12 _ZN9MathWorks3lxe12IRTreeWalker9applyTermEPN2ir4termE + 98
	24  libmwm_lxe.dylib                    0x00000001136371df _ZN9MathWorks3lxe12IRTreeWalker11getNextIRPCENS0_9irpc_typeE + 559
	25  libmwm_lxe.dylib                    0x0000000113638529 _ZN9MathWorks3lxe13IrInterpreter9tree_walkERKNS0_9irpc_typeES4_ + 185
	26  libmwm_lxe.dylib                    0x00000001136383f6 _ZN9MathWorks3lxe13IrInterpreter9InterpretEv + 86
	27  libmwm_lxe.dylib                    0x0000000113718d13 _ZN9MathWorks3lxe15ExecutionDriver9ThreadRunEv + 3283
	28  libmwm_lxe.dylib                    0x000000011371c5f0 _ZN9MathWorks3lxe12IrEngineImpl5ApplyERKN2ir6IrTreeEPvS6_ + 352
	29  libmwm_lxe.dylib                    0x0000000113994c13 _ZN9MathWorks3lxe17IrEngineDecorator5ApplyERKN2ir6IrTreeEPvS6_ + 211
	30  libmwm_lxe.dylib                    0x0000000113aa7c00 _ZN9MathWorks3lxe27FileBasedMatlabCodeFunction16DispatchFunctionEiPP11mxArray_tagiS4_ + 848
	31  libmwm_dispatcher.dylib             0x000000011277c47c _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_ + 796
	32  libmwm_dispatcher.dylib             0x000000011277c8f3 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_ + 35
	33  libmwmcos_impl.dylib                0x0000000119942d32 _ZNK4mcos17MScopedMethodInfo6invokeEPKNS_9COSClientERiPNS_8COSValueEiPKS5_ + 530
	34  libmwmcos_impl.dylib                0x00000001198db37f _ZN24OpaqueMethodCallMetaData11callMMethodEPN4mcos13COSMethodInfoEPP11mxArray_tagPKNS0_9COSClientE + 463
	35  libmwmcos_impl.dylib                0x00000001198de6d9 _ZN24OpaqueMethodCallMetaData23callMethodOrConstructorEP11mxArray_tagbiPS1_iPPKS0_bPKN4mcos9COSClientEPS9_PPNS6_13COSMethodInfoEb + 265
	36  libmwmcos_impl.dylib                0x00000001198df02a _ZN20OpaqueMethodMetaData23callMethodOrConstructorEP11mxArray_tagbiPS1_iPPKS0_bPKN4mcos9COSClientEPS9_PPNS6_13COSMethodInfoEb + 394
	37  libmwmcos_impl.dylib                0x00000001198dc8b8 _Z18mcosCallMethodImplN4mcos9COSNameIDES0_PKNS_9COSClientEiPP11mxArray_tagiPPKS4_bb + 200
	38  libmwmcos_impl.dylib                0x00000001198dd684 _ZN4mcos4impl12omCallMethodENS_9COSNameIDES1_PKNS_9COSClientEiPP11mxArray_tagiPPKS5_b + 84
	39  libmwmcos_impl.dylib                0x000000011996eeb6 _ZN4mcos4impl18omDirectCallMethodERKNS_9COSNameIDES3_S3_iPP11mxArray_tagiPPKS4_ + 134
	40  libmwmcos_impl.dylib                0x00000001199940ba _ZN4mcos22COSGenericDynamicPropIC2ENS_9COSStringENS_9COSNameIDES2_ + 4842
	41  libmwmcos.dylib                     0x0000000110bf62ca _Z18omDirectCallMethodRKN4mcos9COSNameIDES2_S2_iPP11mxArray_tagiPPKS3_ + 58
	42  libmwm_lxe.dylib                    0x0000000113b448c0 _ZN9MathWorks3lxe16MCOSFieldManager10CallMethodEiiPPK11mxArray_tag + 144
	43  libmwm_lxe.dylib                    0x0000000113b5e15e _ZNK9MathWorks3lxe21MCOSMethodParenRplace9GetFirstNEm + 478
	44  libmwm_lxe.dylib                    0x0000000113b4e713 _ZNK9MathWorks3lxe28MxArrayOwningRplaceDecorator9GetFirstNEm + 19
	45  libmwm_lxe.dylib                    0x0000000113b4e713 _ZNK9MathWorks3lxe28MxArrayOwningRplaceDecorator9GetFirstNEm + 19
	46  libmwm_lxe.dylib                    0x0000000113b73643 _ZNK9MathWorks3lxe18RplaceToMultiplace9GetFirstNEm + 19
	47  libmwm_lxe.dylib                    0x0000000113b83ccf _ZN9MathWorks3lxe17at_rparen_nargoutEPvNS_2ts4TypeEPKvS1_S1_ + 511
	48  libmwm_lxe.dylib                    0x0000000113bfce05 _ZN9MathWorks3lxe38cpp_primitive_template_fn_with_contextIXadL_ZNS0_17at_rparen_nargoutEPvNS_2ts4TypeEPKvS2_S2_EEEEbS2_S4_S6_S2_S2_ + 21
	49  libmwm_lxe.dylib                    0x000000011363b177 _ZN9MathWorks3lxe13IrInterpreter17ApplyCppPrimitiveEPN2ir4termES4_RKNS_2ts4TypeES8_ + 407
	50  libmwm_lxe.dylib                    0x000000011363a36f _ZN9MathWorks3lxe13IrInterpreter31process_application_of_constantEPN2ir4termE + 1999
	51  libmwm_lxe.dylib                    0x0000000113638838 _ZN9MathWorks3lxe13IrInterpreter13applyConstantEPN2ir4termE + 248
	52  libmwm_lxe.dylib                    0x0000000113636d12 _ZN9MathWorks3lxe12IRTreeWalker9applyTermEPN2ir4termE + 98
	53  libmwm_lxe.dylib                    0x00000001136371df _ZN9MathWorks3lxe12IRTreeWalker11getNextIRPCENS0_9irpc_typeE + 559
	54  libmwm_lxe.dylib                    0x0000000113638529 _ZN9MathWorks3lxe13IrInterpreter9tree_walkERKNS0_9irpc_typeES4_ + 185
	55  libmwm_lxe.dylib                    0x00000001136383f6 _ZN9MathWorks3lxe13IrInterpreter9InterpretEv + 86
	56  libmwm_lxe.dylib                    0x0000000113718d13 _ZN9MathWorks3lxe15ExecutionDriver9ThreadRunEv + 3283
	57  libmwm_lxe.dylib                    0x000000011371c5f0 _ZN9MathWorks3lxe12IrEngineImpl5ApplyERKN2ir6IrTreeEPvS6_ + 352
[... 78 lines cut ...]
	136 libmwmcr.dylib                      0x0000000112635873 _ZN3mcr7runtime17InterpreterThread4Impl26invocation_request_handlerEl + 35
	137 libmwiqm.dylib                      0x000000011250f390 _ZN3iqm23registerIqmPpeConditionEv + 3232
	138 libmwiqm.dylib                      0x00000001124f912e _ZN5boost6detail8function21function_obj_invoker0INSt3__18functionIFNS_3anyEvEEES5_E6invokeERNS1_15function_bufferE + 30
[... cut ...]
	151 libsystem_pthread.dylib             0x00007fff8938891a _pthread_body + 0
	152 libsystem_pthread.dylib             0x00007fff89386351 thread_start + 13
)
```

This seems related to how irrlitch manages the menubar in macOS. Furthermore it seems they still use the Carbo lib, deprecated some ages ago.

Related references:
http://irrlicht.sourceforge.net/forum/viewtopic.php?p=180371#p180371
https://github.com/zaki/irrlicht/blob/2b18a9ae1926ef9ee473ccaafd51398fd417abb9/source/Irrlicht/Irrlicht.cpp

francesco-romano, 2016-08-29:
Probably we are misusing it.
I tried to create a simple Cocoa App, and even with my "stupid" modification, it crashes everytime.
I think we have to find a way to specify the window handler.

kjyv, 2016-09-16:
Shouldn't Irrlicht creating the window be fine? Why is there already a menubar existing?
In any case, it is possible to pass a window id to irr::createDeviceEx(irrDevParams) by setting irrDevParams.WindowId. This seems to mean that a few other things have to be done manually.
Opening the window in fullscreen works ok.

http://irrlicht.sourceforge.net/docu/structirr_1_1_s_irrlicht_creation_parameters.html#af287810d910a23f8f7db98cef87b6eae

francesco-romano, 2016-11-03:
Hi @kjyv 
I completely missed your answer.
As soon as I have some time I'll give a try again to the library.

traversaro, 2020-02-29:
Recap: if I understood correctly, this is an  inherit limitations of macOS and Cocoa-based OSs,  in the sense that windows can only be created by the main thread, and not by non-main threads,  see also the comment (that in that case applied to VTK) https://github.com/PointCloudLibrary/pcl/issues/253#issuecomment-29716100 .

traversaro, 2020-02-29:
As I think back to this problem, I think the only workaround for this is probably to also provide a network/IPC/RPC wrapper of some kind for all the visualizer API, and in macOS when the visualizer is created spawn a new process (and connect to it) instead of directly create the Window from the non-main thread.

traversaro, 2021-05-17:
cc @nunoguedelha

traversaro, 2022-09-12:
This problem today affected @DanielePucci .

traversaro, 2023-02-20:
Related https://github.com/deepmind/mujoco/issues/742 .