gbionics/idyntree #238: macOS tests are failing on Travis

traversaro, 2016-09-29:
Probably due to an update in the Octave version installed by homebrew.

traversaro, 2016-09-30:
There are actually two problems related to octave 4.2 in homebrew: 
* https://github.com/Homebrew/homebrew-science/issues/4366 , that affected installation of octave and so also the superbuild
* A problem in compiling the bindings, that is probably affecting only the CI of iDynTree .

traversaro, 2016-09-30:
The problems induced by Octave 4.2 are: 
* Octave 4.2 C++ API requires C++11 (fixed in https://github.com/robotology/idyntree/commit/5758a47cf03eaf8b6133e091dc6d29aca00e4a3d).
* RC2 specific problems, induced by having alphabetic characters in the patch number (not worth investigating or solving):
~~~
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:84403:23: error: use of undeclared identifier 'rc2'
  const bool prereq = SWIG_OCTAVE_PREREQ(major, minor, patch);
                      ^
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:779:62: note: expanded from macro 'SWIG_OCTAVE_PREREQ'
  ( (OCTAVE_MAJOR_VERSION<<16) + (OCTAVE_MINOR_VERSION<<8) + OCTAVE_PATCH_VERSION >= ((major)<<16) + ((minor)<<8) + (patch) )
                                                             ^
/usr/local/Cellar/octave/4.2.0-rc2/include/octave-4.2.0-rc2/octave/version.h:35:32: note: expanded from macro 'OCTAVE_PATCH_VERSION'
#define OCTAVE_PATCH_VERSION 0-rc2
                               ^
~~~
* Errors of unknown function in the Octave API, that probably need to be fixed in SWIG itself:
~~~
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2147:5: error: no member named 'op_lshift' in 'octave_value'
    swigreg_binary_op(lshift);
    ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2099:60: note: expanded from macro 'swigreg_binary_op'
if (!octave_value_typeinfo::lookup_binary_op(octave_value::op_##name,tid1,tid2)) \
                                             ~~~~~~~~~~~~~~^
<scratch space>:100:1: note: expanded from here
op_lshift
^
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2147:5: error: no member named 'op_lshift' in 'octave_value'
    swigreg_binary_op(lshift);
    ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2100:57: note: expanded from macro 'swigreg_binary_op'
octave_value_typeinfo::register_binary_op(octave_value::op_##name,tid1,tid2,swig_binary_op_##name);
                                          ~~~~~~~~~~~~~~^
<scratch space>:100:1: note: expanded from here
op_lshift
^
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2148:5: error: no member named 'op_rshift' in 'octave_value'
    swigreg_binary_op(rshift);
    ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2099:60: note: expanded from macro 'swigreg_binary_op'
if (!octave_value_typeinfo::lookup_binary_op(octave_value::op_##name,tid1,tid2)) \
                                             ~~~~~~~~~~~~~~^
<scratch space>:100:1: note: expanded from here
op_rshift
^
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2148:5: error: no member named 'op_rshift' in 'octave_value'
    swigreg_binary_op(rshift);
    ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:2100:57: note: expanded from macro 'swigreg_binary_op'
octave_value_typeinfo::register_binary_op(octave_value::op_##name,tid1,tid2,swig_binary_op_##name);
                                          ~~~~~~~~~~~~~~^
<scratch space>:100:1: note: expanded from here
op_rshift
^
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:84403:23: error: use of undeclared identifier 'rc2'
  const bool prereq = SWIG_OCTAVE_PREREQ(major, minor, patch);
                      ^
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:779:62: note: expanded from macro 'SWIG_OCTAVE_PREREQ'
  ( (OCTAVE_MAJOR_VERSION<<16) + (OCTAVE_MINOR_VERSION<<8) + OCTAVE_PATCH_VERSION >= ((major)<<16) + ((minor)<<8) + (patch) )
                                                             ^
/usr/local/Cellar/octave/4.2.0-rc2/include/octave-4.2.0-rc2/octave/version.h:35:32: note: expanded from macro 'OCTAVE_PATCH_VERSION'
#define OCTAVE_PATCH_VERSION 0-rc2
                               ^
/Users/traversaro/src/codyco-superbuild/build/libraries/iDynTree/bindings/octave/iDynTreeOCTAVE_wrap.cxx:84458:27: error: use of undeclared identifier 'octave_call_stack'
    octave_function *me = octave_call_stack::current();
                          ^
~~~

Given that I don't think we have a lot of  users for the Octave bindings, probably the most convenient thing is to disable octave compilation/test in OS X for the time being (until the proper Octave 4.2 is released and available in homebrew, so at least the second problem will be solved).

traversaro, 2016-12-10:
@nunoguedelha I totally forget about this when discussing about octave in OS X.. did you solved this issues?

nunoguedelha, 2016-12-12:
@traversaro I haven't tried the bindings with octave yet. I should be doing this today or tomorrow. I'm running the 4.2.0 version installed with homebrew.

traversaro, 2016-12-14:
Related : https://github.com/swig/swig/issues/847 .

traversaro, 2016-12-14:
This could be an additional reason to switch on using bindings using the mex API, the same that we use for matlab.

traversaro, 2017-06-15:
Fixed by migrating the mex interface, see https://github.com/robotology/idyntree/issues/305 .

closed 2017-06-15 by traversaro