gbionics/idyntree #242: Parser crashing on malformed URDF

traversaro, 2016-10-26:
@MartaLorenzini Can you attach the offending URDF to this discussion, so we can fix the bug and add it in the tests? Thanks!

MartaLorenzini, 2016-10-26:
[XSensURDF_subj1.txt](https://github.com/robotology/idyntree/files/552883/XSensURDF_subj1.txt)

traversaro, 2016-10-26:
Thanks @MartaLorenzini ! 

The file is this one: 
~~~
<!-- Sensor  1-->
<sensor name="Pelvis_gyro" type="gyroscope">
<parent link="Pelvis"/>
<origin xyz="-0.055797 -6.8455e-05     0.09382" rpy="0.10277     0.99391     -3.0189"/>
</sensor>
<sensor name="Pelvis_accelerometer" type="accelerometer">
<parent link="Pelvis"/>
<origin xyz="-0.055797 -6.8455e-05     0.09382" rpy="0.10277     0.99391     -3.0189"/>
</sensor>
<!-- Sensor  2-->
<sensor name="T8_gyro" type="gyroscope">
<parent link="T8"/>
<origin xyz="0.12904 -5.7446e-05    0.071049" rpy="-0.055985      1.1227   -0.023817"/>
</sensor>
<sensor name="T8_accelerometer" type="accelerometer">
<parent link="T8"/>
<origin xyz="0.12904 -5.7446e-05    0.071049" rpy="-0.055985      1.1227   -0.023817"/>
</sensor>
<!-- Sensor  3-->
<sensor name="Head_gyro" type="gyroscope">
<parent link="Head"/>
<origin xyz="-0.064602  6.0394e-05    0.071069" rpy="2.5761   -0.051806     -1.5359"/>
</sensor>
<sensor name="Head_accelerometer" type="accelerometer">
<parent link="Head"/>
<origin xyz="-0.064602  6.0394e-05    0.071069" rpy="2.5761   -0.051806     -1.5359"/>
</sensor>
<!-- Sensor  4-->
<sensor name="RightShoulder_gyro" type="gyroscope">
<parent link="RightShoulder"/>
<origin xyz="-0.018351   -0.045761   -0.054872" rpy="0.86963   -0.063373    -0.87994"/>
</sensor>
<sensor name="RightShoulder_accelerometer" type="accelerometer">
<parent link="RightShoulder"/>
<origin xyz="-0.018351   -0.045761   -0.054872" rpy="0.86963   -0.063373    -0.87994"/>
</sensor>
<!-- Sensor  5-->
<sensor name="RightUpperArm_gyro" type="gyroscope">
<parent link="RightUpperArm"/>
<origin xyz="-0.018745   -0.093784    0.018795" rpy="-0.43727    0.012838     -1.5425"/>
</sensor>
<sensor name="RightUpperArm_accelerometer" type="accelerometer">
<parent link="RightUpperArm"/>
<origin xyz="-0.018745   -0.093784    0.018795" rpy="-0.43727    0.012838     -1.5425"/>
</sensor>
<!-- Sensor  6-->
<sensor name="RightForeArm_gyro" type="gyroscope">
<parent link="RightForeArm"/>
<origin xyz="0.0017342    -0.18854    0.018831" rpy="-1.1409   0.0066822     -1.4625"/>
</sensor>
<sensor name="RightForeArm_accelerometer" type="accelerometer">
<parent link="RightForeArm"/>
<origin xyz="0.0017342    -0.18854    0.018831" rpy="-1.1409   0.0066822     -1.4625"/>
</sensor>
<!-- Sensor  7-->
<sensor name="RightHand_gyro" type="gyroscope">
<parent link="RightHand"/>
<origin xyz="0.00013779   -0.051904    0.018768" rpy="-1.0793    -0.23419     -1.7229"/>
</sensor>
<sensor name="RightHand_accelerometer" type="accelerometer">
<parent link="RightHand"/>
<origin xyz="0.00013779   -0.051904    0.018768" rpy="-1.0793    -0.23419     -1.7229"/>
</sensor>
<!-- Sensor  8-->
<sensor name="LeftShoulder_gyro" type="gyroscope">
<parent link="LeftShoulder"/>
[... 64 lines cut ...]
</sensor>
<sensor name="LeftUpperLeg_accelerometer" type="accelerometer">
<parent link="LeftUpperLeg"/>
<origin xyz="0.0086013    0.041128    -0.24659" rpy="3.1146      1.3446     -1.6294"/>
</sensor>
<!-- Sensor  16-->
<sensor name="LeftLowerLeg_gyro" type="gyroscope">
<parent link="LeftLowerLeg"/>
<origin xyz="0.032062  -0.0086346    -0.13926" rpy="0.66209      1.4637     0.44871"/>
</sensor>
<sensor name="LeftLowerLeg_accelerometer" type="accelerometer">
<parent link="LeftLowerLeg"/>
<origin xyz="0.032062  -0.0086346    -0.13926" rpy="0.66209      1.4637     0.44871"/>
</sensor>
<!-- Sensor  17-->
<sensor name="LeftFoot_gyro" type="gyroscope">
<parent link="LeftFoot"/>
<origin xyz="0.076661  -0.0013708   -0.010771" rpy="-0.3193     0.63334     0.29874"/>
</sensor>
<sensor name="LeftFoot_accelerometer" type="accelerometer">
<parent link="LeftFoot"/>
<origin xyz="0.076661  -0.0013708   -0.010771" rpy="-0.3193     0.63334     0.29874"/>
</sensor>
~~~
It is highly malformed, because there is no `<robot>` root tag. However the parser should not crash.