gbionics/idyntree #251: Understand why doxygen documentation of some KinDynComputations class methods related to dynamics is not appearing on the website

traversaro, 2016-12-15:
See http://wiki.icub.org/codyco/dox/html/idyntree/html/classiDynTree_1_1KinDynComputations.html#ae5a13daafda43dd17c1def2441a1bba3 vs https://github.com/robotology/idyntree/blob/master/src/high-level/include/iDynTree/KinDynComputations.h#L555 .

giuliavezzani, 2017-01-12:
Can someone assign it to me? I don't have the permission to self-assign it :smile:

traversaro, 2017-01-13:
Done, thanks for the interest! : )

traversaro, 2017-01-13:
Doxygen documentation of idyntree can be produced locally by using the `make dox` command.

giuliavezzani, 2017-01-13:
Thanks for assigning to me! I'll face the problem asap :smiley:

traversaro, 2020-04-06:
This is now working: https://robotology.github.io/idyntree/master/classiDynTree_1_1KinDynComputations.html#a9149e7d3dc280e3fd777998c465c9016 .

closed 2020-04-06 by traversaro