gbionics/idyntree #267: Static analyzer issues

francesco-romano, 2017-03-06:
While running the static analyzer on the iDynTree code, I found some apparently not problematic issues, like the following:

> MatrixDynSize.cpp:108:5: Null pointer argument in call to memory copy function

with the associated stack, e.g.

[image]

The problem is that (taken from documentation of `std::memcpy`)
> If either dest or src is a null pointer, the behavior is undefined, even if count is zero.

cc @traversaro

francesco-romano, 2017-03-06:
@traversaro we are missing the "Volunteers needed" label 🤣

francesco-romano, 2017-03-06:
There are more than one of this kind of (subtle) issues. We should find time to fix them.

francesco-romano, 2017-03-06:
These are the issues found in `idyntree-core`

```

ArticulatedBodyInertia.cpp:192:5: Assigned value is garbage or undefined (within a call to 'operator=')

ArticulatedBodyInertia.cpp:205:5: Assigned value is garbage or undefined (within a call to 'operator=')

ArticulatedBodyInertia.cpp:231:5: Assigned value is garbage or undefined (within a call to 'operator=')

ArticulatedBodyInertia.cpp:249:5: Assigned value is garbage or undefined (within a call to 'operator=')

ClassicalAcc.cpp:41:5: Assigned value is garbage or undefined (within a call to 'operator=')

GeomVector3.cpp:209:9: Assigned value is garbage or undefined (within a call to 'operator=')

InertiaNonLinearParametrization.cpp:178:5: Assigned value is garbage or undefined (within a call to 'operator=')

MatrixDynSize.cpp:108:5: Null pointer argument in call to memory copy function

MatrixDynSize.cpp:236:24: Potential leak of memory pointed to by 'localBuf'

MatrixDynSize.cpp:245:13: Null pointer argument in call to memory copy function

Rotation.cpp:169:9: Assigned value is garbage or undefined (within a call to 'operator=')

RotationRaw.cpp:68:9: Assigned value is garbage or undefined (within a call to 'operator=')

RotationRaw.cpp:78:9: Assigned value is garbage or undefined (within a call to 'operator=')

RotationRaw.cpp:91:9: Assigned value is garbage or undefined (within a call to 'operator=')

RotationRaw.cpp:116:9: Assigned value is garbage or undefined (within a call to 'operator=')

RotationRaw.cpp:130:9: Assigned value is garbage or undefined (within a call to 'operator=')

RotationRaw.cpp:146:9: Assigned value is garbage or undefined (within a call to 'operator=')

SpatialInertia.cpp:138:66: The right operand of '*' is a garbage value (within a call to 'cross')

SpatialInertia.cpp:247:5: The left operand of '-' is a garbage value (within a call to 'compute')

SpatialInertiaRaw.cpp:168:5: Assigned value is garbage or undefined (within a call to 'operator=')

Transform.cpp:413:9: Assigned value is garbage or undefined (within a call to 'operator=')

Transform.cpp:438:9: Assigned value is garbage or undefined (within a call to 'operator=')

Transform.cpp:452:9: Assigned value is garbage or undefined (within a call to 'operator=')

Transform.cpp:486:9: Assigned value is garbage or undefined (within a call to 'operator=')

Transform.cpp:504:9: Assigned value is garbage or undefined (within a call to 'operator=')

TransformDerivative.cpp:102:5: Assigned value is garbage or undefined (within a call to 'operator=')

TransformDerivative.cpp:123:5: Assigned value is garbage or undefined (within a call to 'operator=')

TransformDerivative.cpp:142:5: Assigned value is garbage or undefined (within a call to 'operator=')

TransformDerivative.cpp:166:5: Assigned value is garbage or undefined (within a call to 'operator=')

TransformDerivative.cpp:202:5: Assigned value is garbage or undefined (within a call to 'operator=')

TransformDerivative.cpp:220:5: Assigned value is garbage or undefined (within a call to 'operator=')

```

traversaro, 2017-06-06:
This kind of warnings could be generated by the CI using `scan-build` ( http://clang-analyzer.llvm.org/ ) .