gbionics/idyntree #276: Debug alignement problems arising on 32-bit systems

traversaro, 2017-03-20:
Without the workaround in https://github.com/robotology/idyntree/commit/44f0125e6e38416792ee1dec1980138f9c605bc5 , the InverseKinematics test are failing on 32-bit Windows. We suspect this is due to some aligment issues in some classes in iDynTree. 
To further debug this issue, we plan to run the test suite under Valgrind on Ubuntu 32 bit.

traversaro, 2017-03-21:
The problem can be reproduced on Ubuntu 32-bit, i.e. the tests fail without https://github.com/robotology/idyntree/commit/44f0125e6e38416792ee1dec1980138f9c605bc5 .

traversaro, 2017-03-21:
The problem boils down to the problem that the constructor of `std::vector<std::pair<double, double> > m_jointLimits` is not called, or at least is not called correctly.

traversaro, 2017-03-21:
The problem was actually quite simple: a Vector the element at position 3 was accessed in a vector with size 3 https://github.com/robotology/idyntree/commit/7c3c40dff42ad5767583598014c3836f584dfe48 , and that corrupted the memory of `std::vector<std::pair<double, double> > m_jointLimits`.

A side problem highlighted by this, is that we do not have any check (even in Debug mode) to detect out of range access in `iDynTree::VectorFixSize` and `iDynTree::VectorDynSize`.

closed 2017-03-21 by traversaro

pattacini, 2017-03-21:
Cool 👍