gbionics/idyntree #288: Parsers code is using implicitly assuming that "C" locale is being set

traversaro, 2017-05-05:
In the code of the URDF parsers, we use a number of standard C/C++ functions for conversion from string to double/integer, in particular the [`std::atof`](http://en.cppreference.com/w/cpp/string/byte/atof) function. 

This function works fine, but according to the C/C++ standard its behavior depends on the locale currently set in the process, using the [`std::setlocale`](http://en.cppreference.com/w/cpp/locale/setlocale) function. The problem is that (even if this choice is not actually documented anywhere in the URDF specification or documentation) URDF assume that all the numeric attributes are encoded according to the `"C"` locale, i.e. the default locale setting for C/C++ . The problem is that some libraries such as Qt on some platforms *actually change the process-wide locale to be `""`, i.e. the user-preferred locale, see [1] . 

I noticed this issue because the iDynTree URDF parser was failing to work properly when used in an application using Qt. 

To solve this issue, we need to make sure that each conversion from string to double uses the correct locale, i.e. `"C"` for URDF files. In SDFormat a similar issue [2] was fixed by introducing a call to `std::setlocale` inside the library [3], but I feel that this solution is a prone to error because the on some platforms it modifies the process-wide locale setting, and so it is possible that another thread in the same process sets calls `std::setlocale("")` before the conversion is actually performed. 

The solution that I prefer at the moment is to use an instance of `std::stringstream` for actually performing this numeric conversions for the URDF parser, and to manually set the locale for just that object (that is a local variable of the function) to `"C"` (or more specifical the [`classic()`](http://en.cppreference.com/w/cpp/locale/locale/classic) locale) using the [`imbue(const std::locale& loc)`](http://en.cppreference.com/w/cpp/io/ios_base/imbue) method:
~~~
std::string textToConvert;
double convertedDouble;
std::stringstream ss(textToConvert);
ss.imbue(std::locale::classic());
ss >> convertedDouble;
~~~

[1] http://stackoverflow.com/questions/25661295/why-does-qcoreapplication-call-setlocalelc-all-by-default-on-unix-linux
[2] https://bitbucket.org/osrf/sdformat/issues/60
[3] https://bitbucket.org/osrf/sdformat/pull-requests/147/fix-problems-with-latin-locales-and/diff#Lsrc/Param.ccT145

traversaro, 2017-05-05:
Note that actually checking if the parsing of the number is successful and halting the parser if this is not the case it would make much easier to spot this kind of locale-related problems.

traversaro, 2017-05-05:
Note that for some of the numerical attributes of URDF, the `xsd` file available at specifies that they are of type `xsd::double` [1] . I don't know if format `double` accepted by C++ functions with the 
`"C"` locale is totally compatible with the `xsd::double`, but it would be worth checking. 

[1] https://www.w3.org/TR/xmlschema-2/#double .

traversaro, 2017-05-05:
cc @diegoferigo that is also working on XML parsers.

traversaro, 2017-05-05:
The error in iDynTree classes and functions should be solved by https://github.com/robotology/idyntree/issues/289 . The `urdfdom` library is subject as well to the problem, and it is used in the legacy KDL-based classes. For that we opened an upstream issue in https://github.com/ros/urdfdom/issues/98 .

diegoferigo, 2017-05-08:
Oww this is really nasty, thank for this debugging, it'll bear in mind for the XML parser (https://github.com/robotology-playground/xsens-mvn/pull/12) we'll use for `mvnx` files.

traversaro, 2017-07-14:
Problem solved in the iDynTree's parser, and the urdfdom-parser are in the process of being deprecated, and the upstream issue have been opened. Closing.

closed 2017-07-14 by traversaro

traversaro, 2017-10-18:
Related (old) YARP issue: https://sourceforge.net/p/yarp0/bugs/52/ .

traversaro, 2018-07-04:
Related stackoverflow question: https://stackoverflow.com/questions/13919817/sscanf-and-locales-how-does-one-really-parse-things-like-3-14 .

traversaro, 2018-07-25:
Related tinyxml2 issues: 
* https://github.com/leethomason/tinyxml2/issues/160
* https://github.com/leethomason/tinyxml2/pull/382

traversaro, 2018-09-03:
Updated the original issue with some new alternative solutions.

traversaro, 2019-09-26:
Related PR: https://github.com/fkanehiro/openhrp3/pull/144 .

traversaro, 2020-11-19:
Related issue: https://github.com/ignitionrobotics/ign-rendering/issues/136 .

traversaro, 2020-12-07:
Related comment: https://github.com/JuliaComputing/llvm-cbe/pull/83/files#r537553875 .

traversaro, 2021-02-17:
An interesting comparison of string to double routines: https://github.com/shaovoon/floatbench .

traversaro, 2021-03-18:
Related issue: https://github.com/robotology/gym-ignition/issues/301 .

traversaro, 2022-02-05:
Another occurence of a similar bug: https://github.com/deepmind/mujoco/issues/131 .

traversaro, 2022-11-11:
Related issue on rapidcsv: https://github.com/d99kris/rapidcsv/issues/102 .

traversaro, 2023-10-06:
Some more issues for my collection:
* https://github.com/gazebosim/gz-common/issues/233
* https://github.com/gazebosim/gz-rendering/issues/136
* https://github.com/gazebosim/gz-math/issues/215

traversaro, 2024-04-24:
More issues for my collection:
* https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/723
* https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/817
* https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/817#issuecomment-1923902423

traversaro, 2024-05-14:
Even on Twitter:
* https://x.com/UFOtekkie/status/1790124088100258251

traversaro, 2024-06-05:
https://github.com/google-deepmind/mujoco/issues/845#issuecomment-2150004802