gbionics/idyntree #306: Problem in creating reduced model with FT sensors

traversaro, 2017-05-30:
@S-Dafarra was trying to load the wholeBodyDynamicsDevice with a reduced model for iCub (model: https://github.com/robotology/yarp-wholebodyinterface/blob/master/app/robots/iCubGenova04/model.urdf). For a mechanical problem the right arm was not working, and so he wanted to remove the joints of the right arm. 
Apparently the model reduction was failing because it was not able to deal correctly with the reduction of the [`r_arm_ft_sensor`](https://github.com/robotology/yarp-wholebodyinterface/blob/master/app/robots/iCubGenova04/model.urdf#L2052). 
This seems to be caused by a limitation in the `createReducedModelAndSensors` function, in particular: 
https://github.com/robotology/idyntree/blob/850c0c64d9901e0d448f2d1390a73622dcae2e7b/src/sensors/src/ModelSensorsTransformers.cpp#L45
In fact, it seems that the case in which the FT sensor is attached to some links that are reduced is missing. To fix the problem, it should be necessary to change the FT sensor parameter to be attached to the new link to which the neighbor link have been reduced, as it is done for link sensor in the following lines (see https://github.com/robotology/idyntree/blob/850c0c64d9901e0d448f2d1390a73622dcae2e7b/src/sensors/src/ModelSensorsTransformers.cpp#L104).

traversaro, 2017-05-31:
Related issues/PRs (for @diegoferigo) : 
 * https://github.com/robotology/idyntree/issues/30
 * https://github.com/robotology/idyntree/issues/230
 * https://github.com/robotology/idyntree/issues/234
 * https://github.com/robotology/idyntree/pull/257
 * https://github.com/robotology/idyntree/pull/154
 * https://github.com/robotology/idyntree/pull/103

traversaro, 2017-05-31:
It would be good to add a test for this in https://github.com/robotology/idyntree/blob/850c0c64d9901e0d448f2d1390a73622dcae2e7b/src/sensors/tests/SensorsListUnitTest.cpp (or even in a new file).

traversaro, 2017-05-31:
Generic developer documentation: https://github.com/robotology/idyntree#developer-documentation .

DanielePucci, 2017-07-05:
This issue is also of some importance, update on this @diegoferigo ?

traversaro, 2018-06-27:
Fixed in https://github.com/robotology/idyntree/pull/325 .

closed 2018-06-27 by traversaro