gbionics/idyntree #308: Indeces is not english

diegoferigo, 2017-06-01:
I know it's grammar nazi, but in iDynTree code there are [59 occurrences](https://github.com/robotology/idyntree/search?utf8=%E2%9C%93&q=indeces&type=) of `indeces`, that is a weird mix between `indices` and `indexes`, both correct. Am I missing something?

traversaro, 2017-06-01:
My bad, and I don't think this is being grammar nazi, thanks for the report.

traversaro, 2017-06-01:
I think it is a good occasion to cleanup the deprecation workflow currently in place. 
* Move the current macros https://github.com/robotology/idyntree/blob/2470a71cee49ee3a611195e36b8dc447eca9f154/src/yarp-kdl/include/iCub/iDynTree/DynTree.h#L12 in Utils.h and rename them
* Rename all public methods including `indeces` and leave the old one as deprecated
* Find a way to "deprecated" Indeces.h (that we can afterwords use also to solve https://github.com/robotology/idyntree/issues/132)

closed 2017-06-06 by diegoferigo